adafruit/Adafruit-GFX-Library #202: SPI broken with M0+HX8357+WiFi, works with prior version

ronguest, 2019-03-10:
A long running Feather M0 project needed a few tweaks this week but when I re-built it the display wouldn't update. After much experimenting I determined display updates wouldn't occur if WiFi was used. After some gnashing of teeth I thought to re-build it with prior versions of GFX and HX8357 and when I did the project runs as expected. The combination of GFX 1.4 and HX8357 1.0.7 work fine. The current versions of GFX and HX8357 exhibit the problem where using WiFi interferes with all display updates both in my project as well as in the example code.I notice recent commit mentioning SPI changes so imagine the culprit lies there. I'm not very familiar with SPI handling and so I haven't tried to dig into the changes and do a pull request. Hopefully you can reproduce it using the code below.

I've posted test case code built using Arduino IDE 1.8.8 with Adafruit Feather M0 as the target to illustrate the problem. Please see the **second** code snippet in this [fourm post](https://forums.adafruit.com/viewtopic.php?f=47&t=148727). (The first code snippet used ImageReader and can't be used with the prior versions of the libraries).

ladyada, 2019-03-10:
@PaintYourDragon

MrBryonMiller, 2019-03-16:
I had same problem - here's how I fixed it : 
in Adafruit_SPITFT.cpp I changed the function 'startWrite' 
from : 
`void Adafruit_SPITFT::startWrite(void) {
    if(_cs >= 0) SPI_CS_LOW();
    SPI_BEGIN_TRANSACTION();
}
`
to : 
`void Adafruit_SPITFT::startWrite(void) {
    SPI_BEGIN_TRANSACTION();
    if(_cs >= 0) SPI_CS_LOW();
}
` 
This ordering of SPI_BEGIN_TRANSACTION vs SPI_CS_LOW now matches the ordering used in earlier Library.  

@ronguest can you try this and confirm it takes care of your issue?

PaintYourDragon, 2019-03-16:
I put this change into the 1.4.4 release, please let me know if that clears it up for everyone.

MrBryonMiller, 2019-03-17:
@PaintYourDragon Works for me! Thanks.  Still be good to hear from OP.

ronguest, 2019-03-17:
PlatformIO still won't install 1.4.4 via the build process. It can take 24 hours or longer for it to update for some reason. I pinged the developer. If he responds he has pushed 1.4.4 through then I'll test later today. Otherwise it will presumably be ready tomorrow.

ronguest, 2019-03-17:
Unfortunately there is still a problem with 1.4.4. I received the workaround for PIO and rebuilt the project. The good news is that it no longer completely hangs. The bad news is that BMP do not draw correctly. In the first image attached the icon is drawn properly. In the second attached image there are red vertical lines spaced across the icon. I've never seen this problem before. The working version is built again GFX 1.4 and HX8357 1.0.7. 
[image]
[image]

MrBryonMiller, 2019-03-17:
@ronguest I just want to be sure what you meant when you said "I received the workaround " are you talking about the workaround I proposed or the actual 1.4.4 release?  I ask this because when I made my edits I also had streaking like you show.  I figured I would open a separate issue once I heard from @PaintYourDragon so I did not say anything.  But when I installed the actual release my screen was clean.  Seems 1.4.4 cleaned up many other things besides the ordering issue.

ronguest, 2019-03-17:
Sorry I wasn't clear. I mean I got the workaround for PIO so it would directly grab the latest 1.4.4 from Github. I did cleans on the project and rebuilds with both with the older version of the two libraries as well as with the latest version of both libraries. I did it three times and the result was reproducible, namely no red lines with the old version and red lines with the latest. 

Are you using any libraries in addition to GFX? Below are the libraries in my project:

> lib_deps =
    SD
	https://github.com/adafruit/Adafruit-GFX-Library.git
	https://github.com/adafruit/Adafruit_HX8357_Library.git
	WiFi101
	Adafruit STMPE610

MrBryonMiller, 2019-03-17:
Hmmm, I'm using same libs except I don't use STMPE510 and I do use SPI.h.  I don't think my usage of SPI.h would do anything like this but maybe some issue with STMPE610??  Can you try without using touch?

Also can you try the sketch you referred to from the forum in your initial posting?  I am able to use that without issue.

PaintYourDragon, 2019-03-18:
Second code snippet is working OK here. I could try the other one if the BMPs are available somewhere.

Hardware:
Feather M0 WiFi
3.5" TFT  FeatherWing

Software:
Arduino IDE 1.8.8 Mac
Default SD library
Adafruit_GFX 1.4.7 (DMA NOT enabled in Adafruit_SPITFT.h)
Adafruit_HX8357 1.1.0
Adafruit_ImageReader 1.0.7 (not used in this example, but mentioning it for when I test with BMPs)

PaintYourDragon, 2019-03-18:
In the interim I’m using a couple of moon BMPs from:
https://learn.adafruit.com/wifi-weather-station-with-tft-display/software
These render correctly when using the hardware & software described above and the second code snippet from the Adafruit forums. If you can provide an alternate example (including graphics file(s) if necessary) that could be helpful in troubleshooting this.

ronguest, 2019-03-18:
Thanks for the follow ups. I’ve been tied up on a difficult project the past couple of days and haven’t had time to retrieve the M0 box and re-test. Hoping to grab it in the morning and have some time tomorrow to retest the example with the Arduino IDE. I had wanted to update the project to use the new Image Reader library rather than using the old code I have but the SPI issue blocked me. It would be nice to switch to that code base for the BMP if everything looks ok.

ronguest, 2019-03-19:
Both of the examples in my forum post work correctly so I recommend closing this issue. Thanks for the fix! My project still exhibits the red striping in the icons however. It still works fine with the older libraries. If I think there is a library issue I'll file a separate issue, hopefully with a test case.

Note I tested with GFX 1.4.7 and HX8357 1.1.0.

ronguest, 2019-03-19:
FWIW - apparently the red stripes are due to some issue in the legacy drawBMP code in my project. I replaced it with the new ImageReader library and the stripes are gone. Whee!

closed 2019-03-19 by ronguest

PaintYourDragon, 2019-03-19:
Thanks for testing! Glad it’s resolved.