adafruit/Adafruit-GFX-Library #203: unrestricted unions only available with -std=c++11

cchandel, 2019-03-15:
Hi,
   On Arch Linux, with gcc 8.2.1+20181127-1, Arduino IDE 1.8.8, Adafruit_GFX_Library v 1.4.3, I get this error -

`In file included from /home/cc/Arduino/libraries/Adafruit_GFX_Library/Adafruit_SPITFT.cpp:36:0:
/home/cc/Arduino/libraries/Adafruit_GFX_Library/Adafruit_SPITFT.h:446:9: error: member 'Adafruit_SPITFT::<anonymous union>::<anonymous struct> Adafruit_SPITFT::<anonymous union>::tft8' with constructor not allowed in union
       } tft8;                      ///< Parallel interface settings
         ^

/home/cc/Arduino/libraries/Adafruit_GFX_Library/Adafruit_SPITFT.h:446:9: note: unrestricted unions only available with -std=c++11 or -std=gnu++11`

  Any ideas?

yushuiqiang, 2019-03-18:
I have the same problem.Please help!

PaintYourDragon, 2019-03-18:
Fixed in v1.4.7 of the library. Just updated so it may need a little time to percolate.

closed 2019-03-18 by PaintYourDragon