adafruit/Adafruit-GFX-Library #204: adafruit button launches as if "just pressed"

Emmanuel-DUFOUR, 2019-03-25:
I created a fork with the bug fix, please merge

https://github.com/adafruit/Adafruit-GFX-Library/commit/caea34969cc62fa3e3d347657a4fd071c9aa9fbe

on my system "currstate" and "laststate" are initialised as true by default, so that when the button  is launched it answers "true" to "justPressed" triggering unwanted action. These 2 states must be init as false.

ladyada, 2019-04-05:
hiya you need to submit a PR for us to merge

Emmanuel-DUFOUR, 2019-04-06:
I dont know why but I cant, I go to create pull request, then I compare the master with caea349 , and then there is no option to create the pull request

mikeysklar, 2026-10-15:
Resolved with PR #513, which is now merged. Thanks for reporting this!

closed 2026-10-15 by mikeysklar