adafruit/Adafruit-GFX-Library #205: SPI hardware control of DC & CS signals

PaulStoffregen, 2019-03-27:
A few weeks ago, @LadyAda asked my to consider porting the Teensy-specific optimizations from ILI9341_t3 into the SPITFT abstraction layer.  Sorry, didn't save a link.  At the time I was busy with a tough 6-layer PCB layout.  I've now had a chance to read through the latest SPITFT code.  Starting this issue, so we can discuss what's needed to make this happen.

First, please let me say I feel a little weird about writing this.  Normally I *never* ask anyone to make structural changes to well established software.  But you did specifically ask.  While painful now, long term this could pave the way for other boards with SPI hardware controlled CS pin(s) to also be optimized.

Many newer chips have SPI hardware that can automatically control 1 or several CS pins.  Teensy 3.2, 3.5 & 3.6 can control 5 CS pins.  Arduino Due can control 4 CS pins.  The original idea was you could connect these signals to a decoder chips like 74HC138, so 4 programmable signals could allow up to 15 SPI chip selects.  ILI9341 achieves amazing speed by using 2 of these signals, to control *both* DC and CS.  With each byte written to the SPI transmit register, you must also write the state of these signals during that byte transfer, plus a bit specifying whether you will continue writing more bytes (the signals should remain asserted) or this write is your last (the hardware should auto-deassert the signals).  Because the SPI hardware has a small FIFO buffer, where each word in the FIFO also carries these 5 bits, eliminating digitalWrite() is only a small part the speedup.  The main effect comes from returning before the SPI hardware transmits, which allows GFX to work on the next byte, for nearly perfect 100% SPI bandwidth utilization.

I see 3 obstacles to implementing this in SPITFT.

1: The SPITFT write abstraction doesn't distinguish between your last write (where the signals must be de-asserted) versus continuing writes (where the signals must remain).  For example, where SPITFT currently has writeCommand(byte), ILI9341_t3 has writecommand_cont(byte) and writecommand_last(byte).

2: Two functions, setAddrWindow() and pushColor() are "open ended".  Comments in the code say pushColor() is deprecated, which is good, because it's fundamentally incompatible.  The setAddrWindow() should probably be made private or protected to use only from within SPITFT.

3: Public functions for SPI access should probably be made private or protected from access by Arduino sketches.  While this may be painful for example code and customer projects that have used those functions, long-term to make the GFX and SPITFT abstractions actually an abstraction of the hardware, user-level code needs to be prevented from calling the lowest level hardware access functions directly.

My guess is any of these changes will be too painful?

ladyada, 2019-03-29:
ok let start with

_1: The SPITFT write abstraction doesn't distinguish between your last write (where the signals must be de-asserted) versus continuing writes (where the signals must remain). For example, where SPITFT currently has writeCommand(byte), ILI9341_t3 has writecommand_cont(byte) and writecommand_last(byte)._

do we need a tri-state flag (assert, command, de-assert) or can we have a binary flag (assert, command) and (command, de-assert). i guess there's never any harm in multiple asserts.

PaintYourDragon, 2019-03-30:
> The setAddrWindow() should probably be made private or protected to use only from within SPITFT.
> Public functions for SPI access should probably be made private or protected from access by Arduino sketches.

Very painful. We use both of these as a catch-all for special cases that the library doesn’t handle. Horizontal or vertical mirroring of the display, or pretty much any of the SAMD DMA demos. Removing one or both would require rewriting a volume of code I’d rather not think about (e.g. pretty much everything on HalloWing).

If it’s of any help, some recent changes in SPITFT made it so setAddrWindow() is only called when there’s pixel data to follow (i.e. if there’s clipping or any other zero-pixel situations, the address window just isn’t even set), so there’s probably little harm in leaving it open-ended (though it’s possible there’s some user code out there not doing this).

PaulStoffregen, 2019-03-30:
Yeah, that's the tough part of trying to turn a widely used API of thin hardware functions into an abstraction layer... there's almost always a legacy of code using the thin functions with plenty of hardware assumptions.

Until all that legacy code can be rewritten to use buffers of pixel data and call actual hardware abstraction functions (which I realize is probably never going to happen), I really do not believe attempting to port a different mode of hardware access to SPITFT makes sense.  It would only give customers cases that don't work properly, which is worst than working slowly.

PaulStoffregen, 2019-04-07:
Closing this issue.  Probably not feasible to bring this code into GFX when so many important libs are using GFX as thin functions rather than hardware abstraction.

ladyada, 2019-04-07:
@PaulStoffregen heya ive been swamped all week but i think theres some middle ground where we get 80% of the functionality/speed of DMA with a layer of abstraction that is adaptable to other chips as they come along... whatcha think?

PaulStoffregen, 2019-04-07:
Use of pushColor() by from outside GFX means we can't know when to continue vs end the hardware controlled signals.

ladyada, 2019-04-07:
ok what would be a generic function we would define for blitting?

PaulStoffregen, 2019-04-07:
The problem is GFX doesn't abstract access enough.  By providing setAddrWindow() and pushColor(), GFX lets other libs essentially get very direct access to the hardware - without going though an abstraction layer that would allow doing things differently for FIFO+CS hardware.

I understand you want a hardware abstraction library that can accommodate future hardware.  But as things are now, GFX is a hardware abstraction which allows other libraries to cheat the abstraction layer and gain very direct access.  GFX simply isn't entirely an abstraction layer.  :(

ladyada, 2019-04-07:
yes understood. please be specific about the abstraction you think would work

PaulStoffregen, 2019-04-07:
GFX already has several variants of drawRGBBitmap().  That seems good enough.  I can't see why another function would be added to the API.

The problem isn't GFX lacking anything.  The problem is FGX has too much.  It gives other libs setAddrWindow() and pushColor(), so they can cheat & gain very direct access without abstracting access.  That's probably more efficient.  I'll let Phil comment on why those have been used, rather than the abstract functions.

PaintYourDragon, 2019-04-07:
Where’s the harm then in making such optimizations in the drawRGBBitmap() functions then and leaving the slovenly setAddrWindow() and writePixels() in there, unoptimized and as-is, for existing slovenly code to continue using in an unoptimized manner?

Those two functions are in there mostly because even with the latest hardware you still can’t rely on having that kind of RAM around. It was exploited as a way of slicing things up into manageable chunks. I don’t know. I didn’t write this. Nobody really wrote it, it’s all coalesced from dozens of pull requests from various people, none of whom are really the architect behind it, no one is, and sometimes stuff just gets added in a solving-an-immediate-need fashion. I was just dealing with a lot of this in another library recently, and as much as I’d love to purge a lot of such cruft, I know there’s a lot of projects out there now relying on that cruft, some of which may be in print and un-fixable, so it’s just “Oh well, guess I’m adopting this two-headed kitten then.” Best I can manage in those situations, while we’ve been better documenting more libraries lately, is to clearly indicate what features or functions are deprecated, indicate the preferred alternatives going forward, and make sure the included examples are brought up to date.

PaintYourDragon, 2019-04-07:
Limor and I were talking about maybe making a function like drawRGBBitmap() that accepts a callback to procedurally fill a span of pixels. It would be relatively straightforward to tuck DMA support in there, and this would eliminate most (maybe even all) of the situations where personally I’ve needed to use setAddrWindow() and writePixels() (mostly smooth graphics demo type stuff). Just one case I can think of where I’d still need to send a command directly to the display…but even that might be handled by adding mirroring options to setRotation().

Limor had asked me to unsubscribe from this and I did, but for some reason Github started notifying me again, not sure why. So here we are.

PaulStoffregen, 2019-04-07:
> Where’s the harm then in making such optimizations in the drawRGBBitmap() functions then and leaving the slovenly setAddrWindow() and writePixels() in there, unoptimized and as-is

The harm is to customers who discover some of Adafruit's displays and libraries don't work with certain boards.  That is, unless you can think of a way to implement those functions on hardware that requires specifying what is to be done with CS & DC during each data write, rather than afterwards.

> Nobody really wrote it, it’s all coalesced from dozens of pull requests from various people, none of whom are really the architect behind it, no one is, and sometimes stuff just gets added in a [...]

Yup, computing history is filled with these sorts of stories, where an attempt to abstract from hardware ran into a brick well of legacy code written with deep hardware assumptions.

I completely agree this is probably un-fixable.  That's why I closed this last night.

pljakobs, 2019-04-07:
@PaulStoffregen excuse my ignorance here (and I confess I truely don't understand the intricacies of DMA FIFOs) but:
my understanding is that some functions that Adafruit_GFX exports don't allow the use of DMA/FIFO because they poke all the way through to the display.
also, I think I understand that the library does require some internal state, telling it where it is with regards to sending data or commands as well as if we're at the start or end of a DMA "set" (for lack of a better word).
Wouldn't it be possible to rewrite the offending functions such that they finish any ongoing DMA and fall back to non DMA? If someone uses it, they're SOL using DMA, but it doesn't break things?
It would be a shame if all this work has gone into the library and now we're stuck with forked code because of a, I assume, rarely used set of legacy functions.
If that was the case, I would strongly advocate for a "2.0" version with better API design. Much of the code could carry over and older sketches could continue to use the older API but we would open the path for the next 10 years. I am well aware that that flies in the face of the prime directive that @PaintYourDragon has written down sometime last year.

PaulStoffregen, 2019-04-07:
Maybe someday I'll have time to get more involved in the API design of Adafruit's libraries?  And even Arduino?

But even if I had limitless dev hours in every day, really not sure if that's even appropriate.  As I mentioned in the first message, I generally avoid proposing anyone make major API or structural changes to well established software.  It's pretty always just too painful.

ladyada, 2019-04-07:
@PaulStoffregen moving from non-transaction SPI to transaction SPI was a little gruesome too, but we survived ;)  you mentioned in another thread you wanted to co-create a DMA abstraction to allow various chips to all interface with hardware that can take advantage of it. This would be a good opportunity to implement it in a practice. 
ignore the 'extra' functions for now. like, pretend they dont exist. 
lets talk about what functions can be DMA's and how we could do that in a cross-platform manner.
it benefits teensy because someone will not have to continuously fork/maintain TFT display drivers as new ones come along

PaulStoffregen, 2019-04-07:
Ok, but later, after Teensy 4.0.  That's a much larger project than merely posting the highly optimized code back into GFX.

ladyada, 2019-04-07:
okidoke - i figured it would be a two-fer (whatever we develop, we'd implement here) good luck with your launch!
i'll levae this open to remind us :)

PaulStoffregen, 2019-04-07:
In the meantime, maybe Adafruit could do some sort of audit of all the libraries using GFX?  Long-term, transitioning to using the abstract drawing functions is going to require addressing legacy code.

PaulStoffregen, 2019-04-07:
FWIW, the upcoming Teensy 4.0 hardware will have a "pixel pipeline" peripheral (or "2D GPU") that copies, scales, color transforms, and/or alpha blends - at 1 pixel per clock cycle at 150 MHz.  Whether it'll be usable in the context of GFX or will require a special library is unknown.  But definitely interesting times ahead...

ladyada, 2019-04-07:
yah we recently did a pretty big sweep of the libraries, they've very 'thin' now, 99% of the logic is in SPITFT, we only handle stuff like init, rotation, and the commands for setting the address (which could be protected)
https://github.com/adafruit/Adafruit_HX8357_Library/blob/master/Adafruit_HX8357.cpp

PaulStoffregen, 2019-04-07:
Here's some of that remaining code from Adafruit_HX8357, using startWrite(), writeCommand(), spiWrite() and endWrite().  This sort of programming style in your libraries is not good for the far future.  But in the short term future, the main problem with spiWrite() is no way to know which call will be the last, until it's too late when endWrite() is called.

```
  startWrite();
  const uint8_t *addr = (displayType == HX8357B) ? initb : initd;
  uint8_t        cmd, x, numArgs;
  while((cmd = pgm_read_byte(addr++)) > 0) { // '0' command ends list
    if(cmd != 0xFF) writeCommand(cmd);       // '255' is ignored
    x       = pgm_read_byte(addr++);
    numArgs = x & 0x7F;
    if(x & 0x80) {        // If high bit set...
      delay(numArgs * 5); // numArgs is actually a delay time (5ms units)
    } else {              // Otherwise, issue args to command...
      while(numArgs--) spiWrite(pgm_read_byte(addr++));
    }
  }
endWrite();
```

PaulStoffregen, 2019-04-07:
If you're *really* serious about hardware abstraction, you really should move towards never using these functions outside of GFX, with the aim of eventually making their private to only with GFX.  Allowing this sort of code in your libs is already painful for the abstraction to target Teensy's very similar hardware.  Just imagine how hard this will make targeting much different hardware, like something hosted on top of a MMU which adds context switches in each of those operations.

ladyada, 2019-04-07:
yeah sure, but that's also only run once, so we would just replace that with a 'dataCommand' call - users never see this code, they just call begin() :)

ladyada, 2019-04-07:
@PaulStoffregen do you think it would ever be beneficial to have those sorts of commands run with DMA? they seem fast and they run rarely

PaulStoffregen, 2019-04-07:
Maybe I'm misunderstanding why you want to have hardware abstraction.  When some future boards or boards+kernels come along, isn't the idea to have all the SPI access logic inside GFX, so you can just port GFX and have all your libraries work on that new hardware?

ladyada, 2019-04-07:
yah totally! im asking, instead of startwrite/endwrite, how about we have a function that's like `sendCommand(commandbyte, databytes, databytelen)` - is there anything you see wrong with that?

PaulStoffregen, 2019-04-07:
Yes, that would be much better.

ladyada, 2019-04-07:
awesome, thanks for the quick reply. we'll implement that and test it across all display libraries

PaulStoffregen, 2019-04-12:
Here's a problem that's come up with another Adafruit display.  This really shows how using these low-level SPI functions in all the display libs is a long-term compatibility headache.

https://forum.pjrc.com/threads/55317-New-adafruit-GFX-1-4-x-libs-doesn-t-work-for-me?p=203121&viewfull=1#post203121

And that's just on Teensy 3.6 at 180 MHz.  The coming generations of microcontrollers will bring 500 MHz to 1+ GHz speeds!

ladyada, 2019-04-12:
thanks for the note - please have them open an issue on the matching github repo!

ladyada, 2019-04-15:
okidoke - thanks to Melissa's intrepid testing/hacking, we've added the sendcommand function to GFX and updated SSD1351 (1.27" and 1.5" color OLED). 

e.g. https://github.com/adafruit/Adafruit-SSD1351-library/blob/master/Adafruit_SSD1351.cpp#L348 https://github.com/adafruit/Adafruit-SSD1351-library/blob/master/Adafruit_SSD1351.cpp#L293 and https://github.com/adafruit/Adafruit-SSD1351-library/blob/master/Adafruit_SSD1351.cpp#L307
setAddrWindow is not `protected` yet since we have to update various examples before we do that.

@PaulStoffregen please check it out and then we can continue to the SSD1331 (0.96" OLED)

ladyada, 2019-05-10:
hiya @PaulStoffregen, we've also done SSD1331 and ST77XX (which should also resolve https://forum.pjrc.com/threads/55317-New-adafruit-GFX-1-4-x-libs-doesn-t-work-for-me?p=203121&viewfull=1#post203121)

next up is HX8357 and ILI9341

ladyada, 2019-05-11:
okidoke - HX8357 and ILI9341 are now updated as well
@PaulStoffregen whats the next steps?

PaulStoffregen, 2019-05-15:
Traveling for San Mateo Maker Faire.  Can't look at this until next week.

ladyada, 2019-05-15:
OK happy travels :)

mikeysklar, 2026-11-14:
The outcome agreed here landed: `Adafruit_SPITFT` now has `sendCommand(cmd, data, len)` and `sendCommand16()`.

The Adafruit drivers send their init sequences through it:

| Library | `sendCommand` calls |
|---|---|
| ILI9341 | 7 |
| HX8357 | 4 |
| ST7735 / ST7789 | 9 |
| SSD1351 | 5 |
| SSD1331 | 39 |

The low-level calls left in these drivers sit in their own `setAddrWindow()`, inside the library.

The Teensy hardware CS and DC path still lives in ILI9341_t3, and we don't plan to port it into SPITFT.

Closing this, thanks @PaulStoffregen for the detailed write-up.

closed 2026-11-14 by mikeysklar