adafruit/Adafruit-GFX-Library #207: ESP32 + GFX/ILI9341 + WiFi Not working

pangodream, 2019-04-05:
Hi all,

first of all, I am not pretty sure if I should open the issue in this repository or in the driver one.
If I load the code below to my ESP32 Dev Board, I get some good test executions (about 3 or 4) but suddenly the screen goes blank and nothing is shown.
If I disable the WiFi part (don't connect) the code works perfectly.
I've tried with other pin combinations, but same result.
The code:
`#include <WiFi.h>
#include "SPI.h"
#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"

#define TFT_DC 4
#define TFT_CS 15
#define TFT_RST 2    
#define TFT_MOSI 23           
#define TFT_CLK 18 
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST);

const char* ssid       = "MyWifi";
const char* password   = "MyWifiPassphrase";

void setup() {
  Serial.begin(115200);
  //Wifi connect
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println("Connected!"); 

  tft.begin();
  Serial.println("ILI9341 Test!"); 
}
void loop(void) {
  
  for(uint8_t rotation=0; rotation<4; rotation++) {
    tft.setRotation(rotation);
    testText();
    delay(200);
  }
}
unsigned long testText() {
  tft.fillScreen(ILI9341_BLACK);
  unsigned long start = micros();
  tft.setCursor(0, 0);
  tft.setTextColor(ILI9341_WHITE);  tft.setTextSize(1);
  tft.println("Hello World!");
  tft.setTextColor(ILI9341_YELLOW); tft.setTextSize(2);
  tft.println(1234.56);
  tft.setTextColor(ILI9341_RED);    tft.setTextSize(3);
  tft.println(0xDEADBEEF, HEX);
  tft.println();
  tft.setTextColor(ILI9341_GREEN);
  tft.setTextSize(5);
  tft.println("Groop");
  tft.setTextSize(2);
  tft.println("I implore thee,");
  tft.setTextSize(1);
  tft.println("my foonting turlingdromes.");
  tft.println("And hooptiously drangle me");
  tft.println("with crinkly bindlewurdles,");
  tft.println("Or I will rend thee");
  tft.println("in the gobberwarts");
  tft.println("with my blurglecruncheon,");
  tft.println("see if I don't!");
}
`

Thanks a lot in advance.

ladyada, 2019-04-05:
hiya, you'd need to figure out what in particular is failing - this issue is too vague for us to debug, and you're using hardware we didn't design or develop so it could be anything

pangodream, 2019-04-05:
Ok.
Closing it then.

Thank You.

closed 2019-04-05 by pangodream