adafruit/Adafruit-GFX-Library #210: please enable to erase written chars/lines with a new line by font background color.

dsyleixa, 2019-04-13:
please enable to erase written chars/lines with a new line by font background color.
Better to get some scattered orphan pixels left than the current shambles when writing 2 lines one after another to the same screen postion.
Writing a line of "white spaces" should (mostly) erase the predecessor line, even if not always perfectly

```
#define  textcolor 0xFFFF
#define  backgroundcolor 0x000

display.setCursor(0,0);
display.print("1234abcd  ", textcolor, backgroundcolor);
display.setCursor(0,0);
display.print("efgh6789  ", textcolor, backgroundcolor);
```

pljakobs, 2019-04-13:
@dsyleixa you may want to have a look at my gauges (https://github.com/pljakobs/arduino_gauges)
It's by no means finished but, depending on what you want to achieve, it could be the simplest way.
Say you want to display a measured value, create a text gauge, place it on your display and then only feed the measured value to it. Everything else is handled for you.

If you don't want to use that, you can always take a clue from the way it does what it does:
It writes text to a canvas object and then copies that to screen. While that's a bit less performant, it has a lot of advantages, including that it doesn't produce flickering

If you want to use the library, please also use the Adafruit_GFX library from my fork as it implements a display() function which is needed to update oleds but not implemented in the typical lcd classes.

dsyleixa, 2019-04-13:
> @dsyleixa you may want to have a look at my gauges (https://github.com/pljakobs/arduino_gauges)
> It's by no means finished but, depending on what you want to achieve, it could be the simplest way. [...]

I agree, it's looking really nice, but admittedly by a BIIIG overhead.
Instead, my request was about a vry simple and very quick way to re-write a line by a new one, even if it's not always perfect for italian or bold styles (but which might by handled by additionally using white spaces then).

pljakobs, 2019-04-13:
I agree, it's probably overkill.
But the "new" type fonts have no concept of background, so adding a function as you said would a) require to introduce a background color and b) to store some size and position of field.
You can use the getTextBounds function in conjunction with the drawFilledRect to draw over the old text.

dsyleixa, 2019-04-13:
> 
> 
> But the "new" type fonts have no concept of background, so adding a function as you said would a) require to introduce a background color and b) to store some size and position of field.
> You can use the getTextBounds function in conjunction with the drawFilledRect to draw over the old text.

that's exactly the point, my request was excactly about to have the concept of text background, like for geometric shapes.
Having always to use getTextBounds is far too cumbersome, a very simple and very quick way like stated is requested

pljakobs, 2019-04-13:
Having a notion of textBackground would also imply that background pixels are drawn, which is not the case, since due to kerning, a character can overlap with the background rectangle of a neighboring character.
I thought about this, when first implementing the font rendering code and found it more logical to use fillRect to clear the actual area.

ladyada, 2019-04-13:
@dsyleixa please submit a PR and we'll take a look

dsyleixa, 2019-04-13:
as stated, kerning is no problem now and then, better to have kerning or scattered orphan pixels off and on  instead of the current  helter-skelter.
OTOH, writing a line of blanks should work as well, rather perfectly:

#define  backgroundcolor 0x0000

setTextBackground(backgroundcolor);
setTextColor(textcolor);

display.setCursor(0,0);
display.print("1234abcd");
display.setCursor(0,0);
display.print("         ");
display.setCursor(0,0);
display.print("efgh6789 ");

dsyleixa, 2019-04-13:
> 
> 
> @dsyleixa please submit a PR and we'll take a look

I'm really sorry, I honestly have no idea how a PR works and will be processed

pljakobs, 2019-04-13:
@dsyleixa kerning is a major issue for what you want to do here.
look at the letters VA - the V encroaches on the space of the A and vice versa, they even overlap vertically. Writing out the background pixels for every character means to look at characters in kerning pairs and thus much more complex code.
Also, printing spaces is no solution either since we do not necessarily know how many we need. In a proportional font (as many of the "new" type fonts are), an M uses much more vertical space than an I, for that reason, the String "MMMMM" runs a lot wider than the string "IIIII" - how many spaces would you want to print? The same number as characters in the original string? five? "     " that's good enough for the 5 I string at best, and there is no guarantee.
Again, the fillRect method is what you want to use. Only you know how much space on screen you have devoted for the field you want to print your string to.
I know it's a bit painful, which is why I came up with the idea of the gauges class.

dsyleixa, 2019-04-13:
I know that spacing thing about "I" and "M" and " ".
But
1st, 
one could try different lengths of white-space lines to cover either text - still 1000x better than the cumbersome getTextBounds plus drawFilledRect.
2nd, 
it should work very simple and very quick, even if not 100% clean for either text-rewrite.
3rd, 
no one is forbidden to use the cumbersome getTextBounds plus drawFilledRect if the results don't apply to the demands.

ladyada, 2019-05-05:
we already have a way to set the text color background
https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.cpp#L1238
https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.cpp#L1055
should work for monospace fonts, at least, just fine

dsyleixa, 2019-05-05:
> 
> 
> we already have a way to set the text color background
> https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.cpp#L1238
> https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.cpp#L1055
> should work for monospace fonts, at least, just fine

yes, I know that,
but to my obeservations it does not override then an old line by a new line though, clearing old text pixels.
e.g.:
```

setTextColor(0xFFFF, 0x0000);
display.setCursor(0,0);
display.print("1234abcd");
display.setCursor(0,0);
display.print("         ");
```
also does not override (erase) the old line by blanks,

whilst
```
display.setCursor(0,0);
display.print("1234abcd");
display.setCursor(0,0);
display.print("efgh6789");
```
sill writes to line over each other just producing output mess.

mikeysklar, 2026-10-29:
Background color for custom fonts is on the README list of changes that will not be merged.

Custom font glyphs can overlap, so there is no clean box to fill.

Clear the area with `fillRect()` first, or draw into a `GFXcanvas1` and copy it with `drawBitmap()` and a background color.

Closing this one.

closed 2026-10-29 by mikeysklar