adafruit/Adafruit-GFX-Library #211: request for OLED display.clearDisplay() and display.display(); also to TFTs like  HX3857 and ILI9341

dsyleixa, 2019-04-13:
hi,
I am writing code which ragets different displays.
for OLED SSD1306 there are the functions
display.clearDisplay() 
and 
display.display(); 

but e.g. for TFT 3.5" (HX3857) and ILI9341 that is not available:

I understand that display.display();  perhaps doesn't make sense for thoses TFTs, but for compatibility reasons it should be available as an "empty" specimen though, just doing nothing.

OTOH, display.clearDisplay()  might be implemented as an alias to
display::fillScreen(0x0000);

ladyada, 2019-04-13:
i think that would be much too confusing for people. please use wrapper functions or #ifdefs!

closed 2019-04-13 by ladyada

dsyleixa, 2019-04-13:
I have to disagree:
it's not supposed to be confusing if the GFX lib uses identical API functions for either display.
Instead, it's confusing that some functions work for some and  don't work for other displays.
Please overthink your statement, it's completely unlogic.

pljakobs, 2019-04-13:
@ladyada I have been asking for a
```
virtual void clearDisplay();
virtual void display();
```
as well, it would make portability between different displays easier and would not introduce, so far as I can see, any issues.

ladyada, 2019-04-13:
well look at a PR, but no changes would be required for downstream libraries.

ladyada, 2019-04-13:
@dsyleixa there are multiple commands that exist for only one display - invert, scroll, and probably others.

dsyleixa, 2019-04-13:
the special commands which work just exceptionally for single displays are not essential, 
but e.g. display.display() is essential for the OLED.
Having that in a code which also should control  ILI9341 or HX3857 (optionally, arbitrarily) makes the code impossible to compile.
So please let's start with the 2 ones I mentioned...

ladyada, 2019-04-13:
@dsyleixa okidoke, please chat w @pljakobs to submit a pr!

dsyleixa, 2019-04-13:
I agree if @pljakobs  submitted a PR, I have no idea how that works but I will be fine with either solution!

pljakobs, 2019-04-13:
@ladyada I have an older PR for this open #127 - not sure if it's clean though.
this is the core change:
https://github.com/adafruit/Adafruit-GFX-Library/pull/127/commits/fb2adf0148ab3c60e7e205fcb8926680acf402d1

ladyada, 2019-04-13:
pj, yah can you rebase or resubmit? we've done major updates so we'd want it to pass travis and play nice with the refactors :) thanks!

pljakobs, 2019-04-13:
@ladyada now you're challenging my git knowledge. I will go and improve ;-)

ladyada, 2019-04-13:
pl, in all honesty - probably easiest to close the current PR and re-submit :)