adafruit/Adafruit-GFX-Library #213: touch screen  init is extremely cumbersome and ambiguous depending on chosen portrait or landscape format

dsyleixa, 2019-04-25:
I am using a Adafruit Feather M4 plus TFT Featherwing HX3857 (3.5") 

The touch screen  init is extremely cumbersome and ambiguous depending on chosen portrait or landscape format.

Once having found the correct set of #defines for
TS_MINX 
TS_MAXX 
TS_MINY 
TS_MAXY 
it works then stable, but just for that setting, not for any different one when changing to either  portrait or landscape format.

So I always have to fiddle around with different settings until I'll coincidentally find the right one to start with, and that fails completely when changing the portrait or landscape modes during runtime:

can't that be made automatically detectable after once having set 
tft.setRotation(mode);
when changing mode arbitrarily in the source code during runtime?

```
// Touch Calibration
/*
#define TS_MINX 3800
#define TS_MAXX 100
#define TS_MINY 100
#define TS_MAXY 3750
*/
#define TS_MINX 100
#define TS_MAXX 3800
#define TS_MINY 3750
#define TS_MAXY 100

 /*
 #define TS_MINX 150
 #define TS_MINY 130
 #define TS_MAXX 3800
 #define TS_MAXY 4000 
*/
```

ladyada, 2019-05-05:
no, the tft display and touchscreen are completely separate. in fact, this isnt even the relevant repo for touchscreen stuff. you can open an issue on the touchscreen library repo. thanx!

closed 2019-05-05 by ladyada

dsyleixa, 2019-05-05:
I thought that touchscrees stuff is a subset of GFX?
just to mention, there is already a GFX touch screen issue:
https://github.com/adafruit/Adafruit-GFX-Library/issues/204

where is the (GFX) touchscreen repo you mean?

makermelissa, 2019-05-05:
The 3.5” TFT FeatherWing uses the STMPE610 touch controller. See this guide for more details: https://learn.adafruit.com/adafruit-3-5-tft-featherwing/adafruit2-resistive-touch-screen

dsyleixa, 2019-05-05:
and which repo to report this issue?

makermelissa, 2019-05-05:
Perhaps https://github.com/adafruit/Adafruit_STMPE610.