adafruit/Adafruit-GFX-Library #222: Please fix compiler warnings with -Wall

marcmerlin, 2019-05-12:
Adafruit git master gives me the following when all warnings on turned on in the arduino GUI
```
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_SPITFT.cpp:902:6: warning: unused parameter 'block' [-Wunused-parameter]
 void Adafruit_SPITFT::writePixels(uint16_t *colors, uint32_t len,
      ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_SPITFT.cpp:902:6: warning: unused parameter 'bigEndian' [-Wunused-parameter]
In file included from /var/local/arduino-1.8.6-pre/portable/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,
                 from /var/local/arduino-1.8.6-pre/portable/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from /var/local/arduino-1.8.6-pre/portable/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Arduino.h:32,
                 from /home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.h:5,
                 from /home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:34:
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp: In member function 'void Adafruit_GFX::drawChar(int16_t, int16_t, unsigned char, uint16_t, uint16_t, uint8_t)':
/var/local/arduino-1.8.6-pre/portable/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:74:85: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     #define pgm_read_dword(addr)            (*reinterpret_cast<const uint32_t*>(addr))
                                                                                     ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:60:42: note: in expansion of macro 'pgm_read_dword'
  #define pgm_read_pointer(addr) ((void *)pgm_read_dword(addr))
                                          ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:1076:43: note: in expansion of macro 'pgm_read_pointer'
         GFXglyph *glyph  = &(((GFXglyph *)pgm_read_pointer(&gfxFont->glyph))[c]);
                                           ^
/var/local/arduino-1.8.6-pre/portable/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:74:85: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     #define pgm_read_dword(addr)            (*reinterpret_cast<const uint32_t*>(addr))
                                                                                     ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:60:42: note: in expansion of macro 'pgm_read_dword'
  #define pgm_read_pointer(addr) ((void *)pgm_read_dword(addr))
                                          ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:1077:39: note: in expansion of macro 'pgm_read_pointer'
         uint8_t  *bitmap = (uint8_t *)pgm_read_pointer(&gfxFont->bitmap);
                                       ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp: In member function 'virtual size_t Adafruit_GFX::write(uint8_t)':
/var/local/arduino-1.8.6-pre/portable/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:74:85: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     #define pgm_read_dword(addr)            (*reinterpret_cast<const uint32_t*>(addr))
                                                                                     ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:60:42: note: in expansion of macro 'pgm_read_dword'
  #define pgm_read_pointer(addr) ((void *)pgm_read_dword(addr))
                                          ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:1161:50: note: in expansion of macro 'pgm_read_pointer'
                 GFXglyph *glyph = &(((GFXglyph *)pgm_read_pointer(
                                                  ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp: In member function 'void Adafruit_GFX::charBounds(char, int16_t*, int16_t*, int16_t*, int16_t*, int16_t*, int16_t*)':
/var/local/arduino-1.8.6-pre/portable/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:74:85: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     #define pgm_read_dword(addr)            (*reinterpret_cast<const uint32_t*>(addr))
                                                                                     ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:60:42: note: in expansion of macro 'pgm_read_dword'
  #define pgm_read_pointer(addr) ((void *)pgm_read_dword(addr))
                                          ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:1353:50: note: in expansion of macro 'pgm_read_pointer'
                 GFXglyph *glyph = &(((GFXglyph *)pgm_read_pointer(
                                                  ^
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp: At global scope:
/home/merlin/Arduino/libraries/Adafruit-GFX-Library/Adafruit_GFX.cpp:1515:6: warning: unused parameter 'i' [-Wunused-parameter]
 void Adafruit_GFX::invertDisplay(boolean i) {
```

ladyada, 2019-05-12:
thanks, if you'd like to try to submit a PR to address it, looks like the ESP8266 compiler is picky!

marcmerlin, 2019-05-12:
Sure @ladyada , although this is kind of crucial code that all your stuff depends on, so if the original author has time to look at it, I'd rather :)
Just to be clear, to make sure it's not a compiler/chip issue, are you aware of the File/Preferences/Compiler Warnings setting? If so, try setting it to 'all' to see if it reproduces the same warnings.
I find that it's safer to code with all warnings as it finds unused variables that are often typos/logic errors as well as indexes that run out of bounds on arrays, which has also found real bugs that had been missed by others before.
I agree that the pgm* warnings look harmless, but you'll also notice
* invertDisplay(boolean i) when i is never used
* Adafruit_SPITFT.cpp:902:6: warning: unused parameter 'block'

ladyada, 2019-05-13:
yah please take a look and see if you spot any bugs. fixing compiler warnings is low priority and as you say, its crucial and a lot depends on it

mikeysklar, 2026-10-29:
Current master builds with no warnings from GFX with `-Wall -Wextra`.

| Core | Board | GFX warnings |
|---|---|---|
| arduino:avr 1.8.7 | Uno | 0 |
| adafruit:samd 1.7.17 | Metro M4 | 0 |
| rp2040:rp2040 6.3.1 | Feather RP2040 | 0 |
| esp32:esp32 3.3.8 | Feather ESP32-S3 | 0 |
| esp8266:esp8266 3.1.2 | HUZZAH | 0 |

Both GFX and SPITFT were built, using the GFXcanvas and mock_ili9341 examples.

Closing this one.

Please reopen with the compiler output if you still see a warning on the current release.

closed 2026-10-29 by mikeysklar