adafruit/Adafruit-GFX-Library #224: (mac) fontconvert.c:22:10: fatal error: 'ft2build.h' file not found

Norfeldt, 2019-06-08:
hi, I've faced the same problem to create the .h file from .ttf file , I'm using MacOS  
after trials i made  this procedures and it works 👍 
1- install brew: link :http://macappstore.org/freetype/
*if you wget don't work with you  get it from this command : brew install wget --with-libressl
2-install freetype2 link : https://opensource.ncsa.illinois.edu/confluence/pages/viewpage.action?pageId=61604031 
3- run command : make fontconvert
4- after downloading the .ttf file  and putting it inside the fonconvert folder run command:
./fontconvert Orgy.ttf  18  > Orgy18.h

_Originally posted by @Waged in https://github.com/adafruit/Adafruit-GFX-Library/pull/96#issuecomment-425973309_

I followed the instructions by @Waged and still get the error 

```
$make fontconvert
fontconvert.c:22:10: fatal error: 'ft2build.h' file not found
#include <ft2build.h>
         ^~~~~~~~~~~~
1 error generated.
make: *** [fontconvert] Error 1
```

pljakobs, 2019-06-09:
Looks to me like you have installed freetype just fine, but not the devel packages. I don't know how the Apple system manages packages, but you need to find the freetype2 devel package and also install that. This will contain the header files necessary to build font convert.

pj

Norfeldt, 2019-06-09:
@pljakobs I did try to follow the instruction on https://opensource.ncsa.illinois.edu/confluence/pages/viewpage.action?pageId=61604031

It did create a lot of files (and the header files), but in the directory where I did the commands. 

In what directory should I do it?

makermelissa, 2019-06-10:
Glad to hear. :)

closed 2019-06-10 by makermelissa