adafruit/Adafruit-GFX-Library #225: getPixel() or readPixel() function

AntonioFromBrazil, 2019-06-09:
Does anybody know how to implement a getPixel() or readPixel() for 9341 TFT controller ? I have an application where I need to save the pixel displayed at x,y location before write a new at that same location... I did not find in GFX library any function to do that...

makermelissa, 2019-06-09:
I couldn't see it in the GFX library either and I do think it would be a useful function. If you want to try implementing it yourself, you can take a look at the data sheet at https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf. I would start with the Memory Read section and go from there. If I were to guess, I would think it would involve setting the row and column using `sendCommand()` and then getting the data back using `readCommand8()`.

AntonioFromBrazil, 2019-06-09:
Ok. Thanks for help. I will take a look at 9341 specs

AntonioFromBrazil, 2019-06-09:
Well... according to https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf what I want is on page 116 (Memory read command) of that doc but unfortunately my skill level is not enough to implement a function for that :(

makermelissa, 2019-06-09:
Ok, well let's leave this open for now so you can be notified of the status for when we can get around to it. :)

makermelissa, 2019-06-09:
Actually, could you please open an issue in https://github.com/adafruit/Adafruit_ILI9341? I think it would be better to implement there.

closed 2019-06-09 by makermelissa

xfdr0805, 2020-06-13:
I need this function to implement save bmp picture feature