adafruit/Adafruit-GFX-Library #230: fix compiler's warnings

JAY-M-ERICSSON, 2019-06-16:
When compiling Adafruit_SPITFT.cpp on AVR boards the [Adafruit_SPITFT::writePixels](https://github.com/adafruit/Adafruit-GFX-Library/blob/3cd97982cdc688948015ddf8a6c4b5e903bc7cf6/Adafruit_SPITFT.cpp#L950-L951) member function does not use block nor bigEndian. 

Suggest to use __attribute__ ((unused)) and  changing the signature to

`void Adafruit_SPITFT::writePixels(uint16_t *colors, uint32_t len,
  __attribute__ ((unused)) bool block, __attribute__ ((unused)) bool bigEndian) {
`

to avoid compiler warnings.

ladyada, 2019-06-16:
sounds good - please submit a PR if you can it would be appreciated! :)

Mark1626, 2020-09-28:
I think this got closed by #275, if not I can raise a PR for this

mikeysklar, 2026-10-29:
Current master builds with no warnings from GFX with `-Wall -Wextra`.

| Core | Board | GFX warnings |
|---|---|---|
| arduino:avr 1.8.7 | Uno | 0 |
| adafruit:samd 1.7.17 | Metro M4 | 0 |
| rp2040:rp2040 6.3.1 | Feather RP2040 | 0 |
| esp32:esp32 3.3.8 | Feather ESP32-S3 | 0 |
| esp8266:esp8266 3.1.2 | HUZZAH | 0 |

Both GFX and SPITFT were built, using the GFXcanvas and mock_ili9341 examples.

Closing this one.

Please reopen with the compiler output if you still see a warning on the current release.

closed 2026-10-29 by mikeysklar