adafruit/Adafruit-GFX-Library #231: Adafruit GFX dummy display

croutor, 2019-06-27:
Hi,

Not an issue, more a change request if you want to.
I've made a small project in order to develop some Adafruit GFX screens on my PC:
[](https://github.com/croutor/Adafruit_GFX_dummy_display)
It's using your library of course + wrapping some calls to the SDL library.
The example.cpp is the entry point of the program. It's providing Arduino like setup() and loop() functions lines 130 and139 respectively.
The result is that you can compile and run this program under Linux and get a virtual display window.
I can give this project to you if you want to.
Thanks for your work!

Vincent H.

dopefishh, 2020-07-15:
Impressive. If only I knew this before I created something similar for `XLib`: https://github.com/dopefishh/Adafruit-GFX-X11/

mikeysklar, 2026-10-21:
Thanks for sharing this, @croutor.
A desktop SDL display is handy, and it works well as its own project.
Closing this one.

closed 2026-10-21 by mikeysklar