adafruit/Adafruit-GFX-Library #234: Adafruit_SPITFT fails on Xmega or AVR8X targets e.g. ATmega4809

prenticedavid, 2019-07-02:


ladyada, 2019-07-02:
we don't have any xmega or 4809's but we're happy to take a PR!

prenticedavid, 2019-07-02:
I posted as an "Issue" because I wanted an opinion about a suitable typedef name and an opinion about a single AVR_SPI_WRITE() macro (and a suitable name).

I will Fork later and offer a Pull Request.

David.

ladyada, 2019-07-02:
sure, you can change the name to `ADAGFX_PORT_t` that way there's no conflict

WillEccles, 2019-07-03:
Ran into this same issue today. The Nano Every won't support this library until this is fixed. I'm almost certainly going to fix it myself tonight and would be happy to PR if @prenticedavid doesn't want to/doesn't have the time to.

prenticedavid, 2019-07-03:
I have already submitted a Pull Request.
My apologies for not updating the "Issues"

@WillEccles,
Please try the PR for yourself.    Can you suggest "better" names?
Is there a more appropriate predefined macro to distinguish AVR8X from XMEGA ?

David.

WillEccles, 2019-07-03:
@prenticedavid I already have implemented your fork in my own project. Before I realized you had already done it, I started doing it myself, and `ADAGFX_PORT_t` is the exact typename I chose myself. I think it's about the best option available, unless you used a namespace around `PORT_t`, which I think is far from optimal in this situation.

As for the macro, I have no suggestions one way or another, I think you did a fine job. Thanks for fixing it for all of us!

WillEccles, 2019-07-03:
By the way, I am not sure if you know this, but on GitHub, if you reference the issue number and say something like `fixes #234` in the PR, GitHub will update the referenced issue(s) and automatically close the issue when the PR is merged. I explicitly mentioned the PR number in my previous message so that there was a link from one to the other and back again for future readers.

Of course, if you already knew that, then carry on!

prenticedavid, 2019-07-03:
Thanks for the tip.   I did not know about the Issue#

ladyada, 2019-07-04:
ok this should be all good now, - please try the new release!

closed 2019-07-04 by ladyada