adafruit/Adafruit-GFX-Library #236: esp826 panics

tablatronix, 2019-07-04:
I have started getting panics on esp8266, and have not been able to reproduce or find the cause.

Anyone have seen this before?

```PHP
Exception Cause: Not found

0x40206d19: yield at ??:?
0x40208d00: Adafruit_SPITFT::writeColor(unsigned short, unsigned int) at ??:?
0x40209bcb: SPIClass::beginTransaction(SPISettings) at ??:?
0x40208e09: Adafruit_SPITFT::writeFillRectPreclipped(short, short, short, short, unsigned short) at ??:?
0x402090bd: Adafruit_SPITFT::fillRect(short, short, short, short, unsigned short) at ??:?
0x402060c5: Print::write(char const*) at ??:?
0x40209f11: Adafruit_GFX::fillScreen(unsigned short) at ??:?
0x4020326f: ShowMenu() at ??:?
0x40104ee8: ets_timer_arm_new at ??:?
0x402074bc: delay at ??:?
0x402074d2: delay at ??:?
0x40203773: setup at ??:?
0x40206cd0: loop_wrapper() at core_esp8266_main.cpp:?
0x401007d5: cont_wrapper at ??:?
0x401000c0: app_entry at ??:?
0x4010f000: ?? ??:0
0x4010f15c: ?? ??:0
0x4000044c: ?? ??:0
0x40000000: ?? ??:0
0x40000010: ?? ??:0
0x40000020: ?? ??:0
0x40000030: ?? ??:0
0x40000040: ?? ??:0
```

tablatronix, 2019-07-04:
Why would a yield() cause a panic or wdt?
If i remove it crash stops..
I am calling in setup, not an interrupt or anything like that..

```C++
#if defined(ESP8266)
        do {
            uint32_t pixelsThisPass = len;
            if(pixelsThisPass > 50000) pixelsThisPass = 50000;
            len -= pixelsThisPass;
            // yield(); // Periodic yield() on long fills
            while(pixelsThisPass--) {
                hwspi._spi->write(hi);
                hwspi._spi->write(lo);
            }
        } while(len);
```

tablatronix, 2019-07-04:
Now my hardware wdt resets, no idea what is causing that, there is no anyc callbacks in my code , could gfx be causing that?

ladyada, 2019-07-04:
🤷‍♀ you can try previous versions of GFX to see if its due to new code

tablatronix, 2019-07-05:
hmm

I rolled back esp and gfx as far as I could without having to rollback Adafruit_ILI9341 also

no change

at a loss now, Ill see what esp8266 know about this
[image]

makermelissa, 2019-07-05:
Maybe make sure you have the latest version of ESP8266 BSP? I know they pushed some breaking changes out recently.

tablatronix, 2019-07-05:
What is `BSP` ? I am using platformio, and have tried staging, stable, and 2.0.4

makermelissa, 2019-07-05:
Board Support Package. Ok, I'm not very familiar with platformio, so maybe it's set up differently than Arduino.

tablatronix, 2019-07-23:
I never could figure this out, It seemed to be some kind memory issue, and gfx write was getting stuck in a large or infinite loop it seemed. Going to close for now.

closed 2019-07-23 by tablatronix

tablatronix, 2019-07-26:
https://github.com/esp8266/Arduino/issues/6283