adafruit/Adafruit-GFX-Library #239: textsize not declared

lylenutting, 2019-07-19:
My Arduino script to display 15 temperatures on my TFTLCD worked fine on <Adafruit_GFX.h>    version  1.5.3 Core graphics library but the upgrade to 1.5.6 resulted in 
The error I receive:

C:\Users\lnutting\Documents\Arduino\libraries\Adafruit_TFTLCD\Adafruit_TFTLCD.cpp: In member function 'void Adafruit_TFTLCD::init()':

C:\Users\lnutting\Documents\Arduino\libraries\Adafruit_TFTLCD\Adafruit_TFTLCD.cpp:115:3: error: 'textsize' was not declared in this scope

   textsize  = 1;

   ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

Here is the Adafruit_TFTLCD.cpp     error line: 115:3
96 void Adafruit_TFTLCD::init(void) {
97
98  #ifdef USE_ADAFRUIT_SHIELD_PINOUT
99     CS_IDLE; // Set all control bits to idle state
100   WR_IDLE;
101   RD_IDLE;
102   CD_DATA;
103   digitalWrite(5, HIGH); // Reset line
104   pinMode(A3, OUTPUT);   // Enable outputs
105   pinMode(A2, OUTPUT);
106   pinMode(A1, OUTPUT);
107   pinMode(A0, OUTPUT);
108   pinMode( 5, OUTPUT);
109 #endif
110
111 setWriteDir(); // Set up LCD data port(s) for WRITE operations
112
113 rotation  = 0;
114 cursor_y  = cursor_x = 0;
115 textsize  = 1;
116 textcolor = 0xFFFF;
117 _width    = TFTWIDTH;
118  _height   = TFTHEIGHT;
119}

ladyada, 2019-07-19:
pleaes open an issue in TFTLCD not here - thanx

mikeysklar, 2026-10-21:
As ladyada noted, this error is in adafruit/Adafruit_TFTLCD, not GFX.
GFX master has no `textsize` member; it uses `textsize_x`/`textsize_y` and `setTextSize()`.
Closing this one.
Please open it in the adafruit/Adafruit_TFTLCD repo if you still see it.

closed 2026-10-21 by mikeysklar