adafruit/Adafruit-GFX-Library #241: setTextColor() , no background color

PCov3r, 2019-08-14:
Hello, 
I tried to set a background color for my text using the second argument of the setTextColor function. However, when I write setTextColor(WHITE, BLUE), there is not blue background.

There is my code :`
#include <MCUFRIEND_kbv.h>
#include <Adafruit_GFX.h>    // Core graphics library
#include <Fonts/FreeMonoBoldOblique12pt7b.h>
#include <Fonts/FreeSansBold12pt7b.h>
#include <Fonts/FreeSansBold18pt7b.h>

#include <SPI.h>
#include "SdFat.h"

// Pin numbers in templates must be constants.
const uint8_t SOFT_MISO_PIN = 12;
const uint8_t SOFT_MOSI_PIN = 11;
const uint8_t SOFT_SCK_PIN  = 13;
//
// Chip select may be constant or RAM variable.
const uint8_t SD_CHIP_SELECT_PIN = 10;

// SdFat software SPI template
SdFatSoftSpi<SOFT_MISO_PIN, SOFT_MOSI_PIN, SOFT_SCK_PIN> SD;

#define BLACK    0x0000
#define BLUE     0x001F
#define RED      0xF800
#define GREEN    0x07E0
#define CYAN     0x07FF
#define MAGENTA  0xF81F
#define YELLOW   0xFFE0 
#define WHITE    0xFFFF

MCUFRIEND_kbv tft

void setup() {
  randomSeed(analogRead(A8));
  SPI.begin();
  Serial.begin(9600);
  tft.reset();
  tft.begin(0x9486);
  tft.setRotation(1);
  Serial.print("Initializing SD card...");
  if (!SD.begin(SD_CHIP_SELECT_PIN)) {
    Serial.println("failed!");
    return;
  }
  Serial.println("OK!");
}

void loop() {
tft.fillScreen(WHITE);
  //showBMP("Menu1.bmp",0,0);
  //tft.fillRect(0,0,480,85,BLACK);
  //tft.fillRect(5,5,470,75,WHITE);*
   tft.setFont(&FreeSansBold12pt7b);
  tft.setCursor(112, 55);
  tft.setTextColor(WHITE, BLUE);
  tft.print("Hello");
}
`

pljakobs, 2019-08-14:
Only the builtin font supports background color rendering. If you need to set a background color, either draw a colored rectangle or, if you want to avoid flickering, use a canvas object.

caternuson, 2019-08-14:
@pljakobs is correct - thanks :). More info:
https://learn.adafruit.com/adafruit-gfx-graphics-library/graphics-primitives#characters-and-text-3-24

> Note that the text background color is not supported for custom fonts. For these, you will need to determine the text extents and explicitly draw a filled rectangle before drawing the text.

closed 2019-08-14 by caternuson