adafruit/Adafruit-GFX-Library #242: SPI incompatibility with nrf52840 (Nano 33 BLE and BLE Sense)

LittleHobbyShop, 2019-08-18:
A difference in SPI.h for the new nrf52840 based boards causes function calls from Adafruit_SPITFT to non-existent functions **setClockDivider()**, **setBitOrder()**, and **setDataMode()**. The result is compile errors.

ladyada, 2019-08-18:
erk that's not good, i think it will fail later on if we don't have transactions or SPI control - your patch doesnt solve the underlying issue :/

LittleHobbyShop, 2019-08-18:
I thought that would be the case, It was a best guess at getting it working with some help on the arduino forum, just wanted to highlight the issue really in the hope someone who knows better can come up with a better solution. This does work though. This is the post i started on the arduino forum [https://forum.arduino.cc/index.php?topic=631882.0](https://forum.arduino.cc/index.php?topic=631882.0)

ladyada, 2019-08-18:
@sandeepmistry hihi id like to get this working for sure, but if SPI transactions or SPI control is coming down the pipe we can wait so its implemented 'right' :) we kinda need transactions because otherwise we get :( when people use displays along with wifi or radio modules

sandeepmistry, 2019-08-19:
Hi @ladyada,

Thanks for letting us know about this, I've opened the following issue to track this: https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/5 - it's in the queue for someone in the Arduino team to look at it.

@LittleHobbyShop please follow the issue I've linked for progress updates on this topic :)

ladyada, 2019-08-19:
OK this patch should not be needed, because the SPI implemenation has transaction support. please update when they have a new BSP and you'll be good to go :D please open a new ticket if it doesnt work then (for some other reason)

closed 2019-08-19 by ladyada