adafruit/Adafruit-GFX-Library #245: Build errors when including from src directory

FrankvdStam, 2019-08-29:
My preferred way of including libraries is from a directory called src in the sketch folder. I do this so that the library is included in my git repo and when doing a clone, the result is the most complete. You really don't depend on libraries from managers or cached files.

Doing this with this library, combined with the ST7735 library, results in build errors that I don't know how to resolve. It's probably relatively simple, since it's a bunch of multiple definition of function errors (function x first defined here). 

[How to include from subfolder of sketch folder](https://forum.arduino.cc/index.php?topic=445230.0)
Jan 01, 2017, 02:59 am
In recent versions of the Arduino IDE(including 1.6.10) if you want to include libraries from the sketch folder you need to put them in a src subfolder. For example:

Blink
|_Blink.ino
|_src
----|_BlinkLib
      ----|_BlinkLib.h

    #include "src/BlinkLib/BlinkLib.h"

I have the same structure with src containing Adafruit_GFX and Adafruit_ST7735_and_ST7789. Sketch contains these lines:

    #include "src/Adafruit_GFX/Adafruit_GFX.h"
    #include "src/Adafruit_ST7735_and_ST7789/Adafruit_ST7735.h"

    void setup() {
      // put your setup code here, to run once:
    }

    void loop() {
     // put your main code here, to run repeatedly:
    }

I'll throw the full build log in a separate comment (it's huge). I target a DOIT esp32 devkit v1, but also a plain arduino will fail. 

For clarity, I'm not asking for a fix in the library's master branch, my way of including might just not be supported, I'm asking for a way to make it work for me.

Cheers.

FrankvdStam, 2019-08-29:
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Frank\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Frank\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Frank\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=10C4_EA60 -ide-version=10809 -build-path C:\Users\Frank\AppData\Local\Temp\arduino_build_451326 -warnings=default -build-cache C:\Users\Frank\AppData\Local\Temp\arduino_cache_311053 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Frank\Desktop\sketch_aug29a\sketch_aug29a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Frank\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Frank\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Frank\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=10C4_EA60 -ide-version=10809 -build-path C:\Users\Frank\AppData\Local\Temp\arduino_build_451326 -warnings=default -build-cache C:\Users\Frank\AppData\Local\Temp\arduino_cache_311053 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Frank\Desktop\sketch_aug29a\sketch_aug29a.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_451326\\sketch\\sketch_aug29a.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Frank\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_451326\\sketch\\sketch_aug29a.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Frank\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src" "C:\\Users\\Frank\\AppData\\Local\\Temp\\arduino_build_451326\\sketch\\sketch_aug29a.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\\Frank\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\S [... cut ...]
C:\Users\Frank\AppData\Local\Temp\arduino_build_451326\sketch\src\Adafruit_GFX\Adafruit_SPITFT.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

Using library Adafruit_GFX_Library at version 1.5.6 in folder: C:\Users\Frank\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
exit status 1
Error compiling for board Arduino/Genuino Uno.

ladyada, 2019-08-29:
hiya this is not supported and the code works when you run it normally. you also have two copies. closing cause this isnt a library issue.

closed 2019-08-29 by ladyada

FrankvdStam, 2019-08-29:
I don't have two copies.

makermelissa, 2019-08-29:
I see 3 copies:
Copy 1: C:\Users\Frank\Documents\Arduino\libraries\Adafruit_GFX_Library
Copy 2: C:\Users\Frank\AppData\Local\Temp\arduino_build_451326\libraries\Adafruit_GFX_Library
Copy 3: C:\Users\Frank\AppData\Local\Temp\arduino_build_451326\sketch\src\Adafruit_GFX

FrankvdStam, 2019-08-29:
Obviously I don't manage the files in temp\arduino_build_xxxxxx, I'd assume the toolchain does that. As for the files in the libraries directory, I deleted them, no effect, re-added them purely to create this issue (reproduction), no effect.

FrankvdStam, 2019-08-29:
Anyone stumbling upon this issue, in Adafruit_ST7735_and_ST7789/Adafruit_ST77xx.h

change

    #include <Adafruit_GFX.h>
    #include <Adafruit_SPITFT.h>
    #include <Adafruit_SPITFT_Macros.h>

to

    //#include <Adafruit_GFX.h>
    //#include <Adafruit_SPITFT.h>
    //#include <Adafruit_SPITFT_Macros.h>

    #include "../Adafruit_GFX/Adafruit_GFX.h"
    #include "../Adafruit_GFX/Adafruit_SPITFT.h"
    #include "../Adafruit_GFX/Adafruit_SPITFT_Macros.h"

To fix the problem.