adafruit/Adafruit-GFX-Library #247: Please support character background painting also for custom fonts

zapta, 2019-09-27:
Hello GFX maintainers,

This request is related to this comment

https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.cpp#L1136

The rationals for supporting background painting also for custom fonts are

1. Custom fonts are important because of the limited size availability of the built in font (no real size scaling, just pixel replication)

2. The comment assumes that custom fonts are not monospace. Custom fonts can be monospace if the user choses so.

3. Even for non monospace fonts, being consistent with the native fonts and painting the background is more intuitive. The user still has the option of setting text_color == bg_color for the 'leave background as is' affect, similar to the native font.

Please reconsider this decision and provide a way to overwrite custom fonts backgrounds. It's much better than the alternative of using the tiny native font or forking this library. 

If you are concerned about backward compatibility, a compile time conditional '#define OVERWRITE_CUSTOM_FONTS_BACKGROUND' should do the trick. 

Thanks.

ladyada, 2019-09-27:
hiya if you'd like you can submit a helper function that will draw the background with a rectangle and then the text on top

zapta, 2019-09-27:
Thanks ladyada@.  I tried it before submitting this request but it flickers. I ended up forking this library and making the custom fonts consistent with the native ones and it works well.  However, forking is evil. Especially for such a basic functionality that will benefit many.

Any reason for not adding a user controlled OVERWRITE_CUSTOM_FONTS_BACKGROUND #define? I see only benefits for GFX from it and it will be backward compatible for old designs.

ladyada, 2019-09-27:
we wont accept #define hacks because people can't use them - you can try to submit a function or helper and we'll take a look

mikeysklar, 2026-10-29:
Background color for custom fonts is on the README list of changes that will not be merged.

Custom font glyphs can overlap, so there is no clean box to fill.

Clear the area with `fillRect()` first, or draw into a `GFXcanvas1` and copy it with `drawBitmap()` and a background color.

Closing this one.

closed 2026-10-29 by mikeysklar