adafruit/Adafruit-GFX-Library #250: Custom fonts for Button labels

donGerry, 2019-10-24:
Hello,
I've created an extension to the drawButton() method to allow the use of custom fonts for the button labels. Without these changes the labels were not placed correctly inside the buttons (i.e. in the center) when setting a custom font.
This extension (files Adafruit_GFX.cpp and Adafruit_GFX.h) are currently commited to my local clone.
Is there a branch where I could push it to, for review and the integration in the library ?

donGerry, 2020-04-15:
Hi, it's me again,
just in case someone is interested in adding my changes to the library,
I've attached my modified files here:
[Adafruit_GFX_Library.zip](https://github.com/adafruit/Adafruit-GFX-Library/files/4481976/Adafruit_GFX_Library.zip)
ciao

ArifBryan, 2022-05-28:
Thank you for your work, it works perfectly!

VictorMarinus, 2023-06-05:
Thanks, this saved my day!

mikeysklar, 2026-10-15:
Resolved with PR #454, which is now merged. Thanks for reporting this!

closed 2026-10-15 by mikeysklar