adafruit/Adafruit-GFX-Library #251: Using setCursor() causes Neopixel to not work (all white)

jkenny23, 2019-11-13:
I've still got some work to do to debug this, like trying in a clean program environment and using an oscilloscope to find out what's going on, but so far I've narrowed down an issue where using an SSD1306 I2C OLED (no reset pin) works fine on its own, and using a pair of Neopixels (WS2812B) using the Adafruit_NeoPixel-ANDnXOR library works fine on its own on an STM32duino Blue Pill, but when I use the setCursor function of the SSD1306 library, the Neopixels are stuck full white. I've done as much searching as I could do, and others have noted some incompatibility but attributed it to running out of memory. I highly doubt that, I used a function to check and I have at least 11kb free (compiler says about the same).

ladyada, 2019-11-13:
hiya this bug report is too vague/complex for us to track down, please use a debugger to figure out where your failure is happening. if you find a specific bug, reopen an issue :)

closed 2019-11-13 by ladyada

jkenny23, 2019-11-14:
Resolved anyways, FYI I think it was a RAM issue. Worked after cleaning up other parts of my code a bit.