adafruit/Adafruit-GFX-Library #255: Library apparently not working on miniTFTWing after version 1.5.7

astrorafael, 2019-12-17:
I've been running the [graphictest example](https://github.com/adafruit/Adafruit-ST7735-Library/blob/master/examples/graphicstest/graphicstest.ino) found in the Adafruit ST7735 and ST7789 Library with the latest version (1.7.0) of Adafruit GFX Library and it doesn't work, the display doesn't show anything.

The example runs perfectly when I switch back to GFX version 1.5.7

Tested on an Arduino Nano with miniTFT wing and the following definitions
```C
#define TFT_RST -1  // miniTFTwing uses the seesaw chip for resetting to save a pin
#define TFT_CS   5 // Arduino Nano D5 pin
#define TFT_DC   6 // Arduini Nano D6 pin
```

caternuson, 2019-12-18:
Looks like it might be affecting ILI9341 also? This is a 2.8" TFT:
https://forums.adafruit.com/viewtopic.php?f=22&t=159912#p787524

makermelissa, 2019-12-18:
Ok, I have confirmed the issue exists with an Adafruit Metro 328 and a 2.8" ILI9341. I'll start there.

makermelissa, 2019-12-18:
The Adafruit Metro 328 and a 2.8" ILI9341 combo work in 1.6.1, so it's a change between 1.6.1 and 1.7.0 causing the issue.

makermelissa, 2019-12-18:
Ok, I think I got it working. Issue was that in sendCommand(), dataBytes was changed from `uint8_t` to `const uint8_t` without adding in pgm_read_byte(). Adding this seems to fix it with that combo. Submitting a PR and hopefully that fixes things.

makermelissa, 2019-12-18:
@astrorafael, I just released version 1.7.1 that should fix this. Could you please try it and let me know? It should take a couple hours to show up in Arduino or you can download it from this repo now.

astrorafael, 2019-12-19:
Yes, it works ! 
However, it seems that the width and height dimensions are swapped in my display
See [this image](https://drive.google.com/open?id=1k6zVFvhaqEJkqzaPj5MJdf91hhF6mVXK)

doublec4, 2019-12-19:
This worked for me as well. Although previously the "graphicstest" example would continuously cycle through all of the graphics displays forever... now it cycles through the graphics once and starts repeating that "groop / deadbeef display" perpetually. No idea if that is related to this update (not that it really even matters since it is just a graphics example) but worth nothing.

ladyada, 2019-12-19:
thats normal, it does a rotation test at the end

ladyada, 2019-12-19:
oops reopening to check the rotation thing

makermelissa, 2019-12-19:
The rotation thing works with 1.6.1, so that must have been another thing that changed between 1.6.1 and 1.7.0.

makermelissa, 2019-12-19:
Ok, looks like there was a total of 3 problems here:
* The const version of sendCommand was not working for 8-bit procs. (fixed)
* The non-cost version of sendCommand was removed causing the rotation issue.
* Width and Height of the MiniTFT were somehow swapped.

I have 2 PRs coming later. One fixes the rotation issue. The other will be in the 7735 and fixes the swapped width and height. I'm unsure what caused that, but it was wrong on my 0.96" MiniTFT display as well and I just swapped them in the initialization, which worked.

makermelissa, 2019-12-19:
PRs are #258 and https://github.com/adafruit/Adafruit-ST7735-Library/pull/101.

astrorafael, 2019-12-20:
I've tested both master branch versions of GFX and ST7735 libraries and the example sketch now runs fine.

makermelissa, 2019-12-20:
Excellent. I published releases on both repos this morning.

closed 2019-12-20 by makermelissa