adafruit/Adafruit-GFX-Library #267: Incorrect library dependency "Adafruit ILI9341"

MHotchin, 2020-01-08:
The library.properties was recently updated to add a dependency to "Adafruit ILI9341".  Although a useful addition, the GFX library is not *dependent* on it, and in fact "Adafruit ILI9341" is already marked as dependent on this library.

ladyada, 2020-01-12:
the examples we use for CI testing are dependent on Adafruit ILI9341
arduino-cli is ok with cyclic deps

closed 2020-01-12 by ladyada

XFactHD, 2020-04-04:
At the same time this dependency completely breaks PlatformIO. It causes it to add the ILI9341 library to the dependency graph. This is problematic when compiling for the Teensy 4 because the TouchScreen library the ILI9341 library depends on does not compile on that platform.

ladyada, 2020-04-04:
@XFactHD please chat with @ivankravets at platformio, we have to put all dependencies in for all examples to test CI, having a dependancy should not break installation process if it is not actually referenced in the code - there are lots of other libraries that have single-platform-only-dependancies!

XFactHD, 2020-04-04:
Ok, I'll do that, thanks.

ivankravets, 2020-04-06:
[PlatformIO Library Dependency Finder](https://docs.platformio.org/en/latest/librarymanager/ldf.html) automatically installs and adds to a build process dependencies declared by a user in a project or by library maintainer. If there are no strict declared dependencies but LDF is enabled (default behavior), PlatformIO will try to resolve dependencies automatically similar to what Arduino IDE does.

Firstly, we respect the project's instruction where Arduino IDE does automatic work for your where sometimes you do need some libraries. In any case, you can [lib_ignore](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore) dependencies which were found by LDF and remove them from the build process.

So, just add `lib_ignore = Adafruit ILI9341`to `plaformio.ini`.

ladyada, 2020-04-06:
thanks @ivankravets ill repeat that when others ask :)

ivankravets, 2020-04-06:
@ladyada Thanks! I saw your photo from the NY office. Everything will be good 👍

XFactHD, 2020-04-06:
> So, just add `lib_ignore = Adafruit ILI9341`to `plaformio.ini`.

That's a great tip, thanks