adafruit/Adafruit-GFX-Library #273: compile error with freetype version 2.10.1

AthusVieira, 2020-02-18:
trying to compile fontconvert but see this error:
warning: implicit declaration of function 'FT_Property_Set'; did you mean 'FT_Face_Properties'? [-Wimplicit-function-declaration]
   FT_Property_Set(library, "truetype", "interpreter-version",
   ^~~~~~~~~~~~~~~
   FT_Face_Properties

Looks like the version 2.10.1 (probably 2.10.0 too) expect the fontconvert.c to include these headers as laided out [here](https://github.com/servo/libfreetype2/blob/master/freetype2/docs/CHANGES#L752)

#include FT_MODULE_H
#include FT_CFF_DRIVER_H

After including these headers, the fontconverter was compiled successfully.

ladyada, 2020-02-18:
hihi please submit a PR :)

BillyDonahue, 2020-05-28:
fixed now. The comment on #297 didn't use the right magic words to autoclose this on merge.

ladyada, 2020-05-28:
rad :)

closed 2020-05-28 by ladyada