adafruit/Adafruit-GFX-Library #274: tft.begin(uint32 freq) does not change frequency

bdlabitt, 2020-02-24:
- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**
Feather M4 Express
- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**
1.8.9
- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

Added two lines of code to graphicstest.  
#define TFT_FREQ 32000000
tft.begin(TFT_FREQ); 

Put oscilloscope probe on SPICLK at the ILI9341 display.  Full code below.
[code]
/***************************************************
  This is our GFX example for the Adafruit ILI9341 Breakout and Shield
  ----> http://www.adafruit.com/products/1651

  Check out the links above for our tutorials and wiring diagrams
  These displays use SPI to communicate, 4 or 5 pins are required to
  interface (RST is optional)
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/

#include "SPI.h"
#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"

// For the Adafruit shield, these are the default.
#define TFT_DC 9
#define TFT_CS 10

// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
// If using the breakout, change pins as desired
//Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);

#define TFT_FREQ 33000000

void setup() {
  Serial.begin(115200);
  while(!Serial) {
    ; //wait for serialport to connect
  }
  
  Serial.println("ILI9341 Test!"); 
 
  tft.begin(TFT_FREQ);

  // read diagnostics (optional but can help debug problems)
  uint8_t x = tft.readcommand8(ILI9341_RDMODE);
  Serial.print("Display Power Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDMADCTL);
  Serial.print("MADCTL Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDPIXFMT);
  Serial.print("Pixel Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDIMGFMT);
  Serial.print("Image Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDSELFDIAG);
  Serial.print("Self Diagnostic: 0x"); Serial.println(x, HEX); 
  
  Serial.println(F("Benchmark                Time (microseconds)"));
  delay(10);
  Serial.print(F("Screen fill              "));
  Serial.println(testFillScreen());
[... 298 lines cut ...]
  unsigned long start;
  int           i, i2,
                cx = tft.width()  / 2 - 1,
                cy = tft.height() / 2 - 1;

  tft.fillScreen(ILI9341_BLACK);
  start = micros();
  for(i=min(tft.width(), tft.height()); i>20; i-=6) {
    i2 = i / 2;
    tft.fillRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(0, i, 0));
    yield();
  }

  return micros() - start;
}
[/code]

CPU Speed 120 MHz
-Os
debug off

Also does not generate correct clock for other higher speed cpu clocks, or higher level of optimization. Expect SPICLK to be set to TFT_FREQ.  Documentation states this will happen.

[image]
SPI clock is 24MHz, not 32MHz

bdlabitt, 2020-02-28:
Sorry, last line should read, "SPI clock is 24MHz, not 33MHz".  SPI clock is not being set to TFT_FREQ.

aklein24, 2020-03-21:
I'm noticing similar behavior on a Particle Boron. Might be unrelated but seems like the same issue.

bdlabitt, 2020-11-23:
I'm beginning to believe this is posted in the wrong place.  If so, where should it go?  ILI9341?  No review or comment in over 6 months.  Not even an acknowledgement of the issue.   From what I can determine, proper DMA SPI management is non-trivial.  Furthermore the solution appears to need to be compatible with SPI.*, SERCOM.*, startup.c, Adafruit_SPIDevice.*, and perhaps Adafruit_ZeroDMA.*

For my application, I am not using the SD card on the TFT board.  So it would be preferable to have a solution with SPI at 40-48MHz for write, and perhaps 24MHz for read for an M4 Express.  By 40-48MHz it is meant a single fixed frequency somewhere around this value.

orhanyor, 2021-04-20:
i was wondering about overclocking SPI speed too. i found this SPI speed related info on here : https://learn.adafruit.com/adafruit-feather-m4-express-atsamd51?view=all 
Look for "Max SPI and Max QSPI" in the article. this is exactly what you are looking for but it seems like adafruit removed the Max SPI option in Arduino, theres only a setting for MAX QSPI.

bdlabitt, 2022-05-05:
Wow, Over 26 months, no response from any developers.  Quite sad state of affairs.  Really disappointed.

Yes, that is where I learned about the possibility to overclock a TFT display.  Apparently no one does it, or the original statement was pure balderdash, or it did work for a while and was removed.

aklein24, 2022-05-05:
@bdlabitt I tried unsuccessfully to increase the SPI clock speed on several different arduinos without success. It's something buried deeper in the IDE than I was able to figure out. I'm disappointed as well.

bdlabitt, 2022-05-07:
Appears my comment was deleted.  Not sure if I crossed a line.  Apologize if I did.  Wasn't intentional bashing.  
Disappointing there is no apparent solution to this issue after 26 months.  A lack of solution is ok, but some sort of explanation why this is difficult or impossible, would be greatly appreciated.

dhalbert, 2022-05-07:
The SPI clock rate setting at high clock rates on the SAMD chips goes by big jumps at high frequencies, because it's a small divisor value.. You may need to try 48MHz instead of 24. See for instance https://github.com/adafruit/circuitpython/issues/464#issuecomment-347676877
and
https://github.com/adafruit/circuitpython/pull/560
That discuss SAMD21 more than SAMD51, but the same problem is present.

bdlabitt, 2022-05-07:
@dhalbert Thank you for a clue!  Haven't jumped to the URLs yet, but notionally how does one change the clock?  More specifically, which clock?  Going off to read...

bdlabitt, 2022-05-07:
Some good discussion in the other threads.  Unfortunately, not being as skilled in the art as many are, it is unclear what to set and where to set it.  A few breadcrumbs would be appreciated.  Thanks!

dhalbert, 2022-05-07:
I  mean that to get a higher SPI frequency than 24MHz, you may need to specify 48MHz, not just 32 or 33MHz. Anything in between will just take the nearest lower possible frequency. 32MHz or 33MHz is probably not possible.

bdlabitt, 2022-05-07:
It is worth a try!  Not sure the display can manage it, but it is worth trying.  So just tft.begin(48000000) ?  Hope to try this today.

dhalbert, 2022-05-07:
yes, try that, or even higher, but you have to change the Arduino menu items, as mentioned, and we don't guarantee it will work.

bdlabitt, 2022-05-07:
"but you have to change the Arduino menu items"
I don't understand this.  Which Arduino menu items and where are they found?  I don't recall seeing any menu items related to SPI.  At this point I'm running 1.8.19 IDE.

dhalbert, 2022-05-07:
Aha, I think the Board Support Package originally had a menu item to adjust the max SPI speed, and that was documented in the guide. However, we took that option away: https://github.com/adafruit/ArduinoCore-samd/pull/110. Instead you can set the clock source, using `SPIClass::setClockSource()`. The possible values are [here](https://github.com/adafruit/ArduinoCore-samd/blob/137b8d2de0e04021d9f656dcf8ea9d4f594b3d1c/cores/arduino/SERCOM.h#L155).

The default clock source is the 48MHz clock, I think. Try 48MHz for now, without changing the clock source. If that doesn't work (24 MHz might be the max for a 48MHz clock), you can try changing the clock source to the CPU frequency (120 MHz) or the 100 MHz clock), so you can get higher frequencies.

bdlabitt, 2022-05-07:
Yes.  The board support package did offer this - but only for a short while!  I remember using it, and then it vanished from the IDE.  Thanks for some ideas to try!  I will try a couple things and get back to you.  Think I will be able to do this this weekend.

bdlabitt, 2022-05-08:
Hmm, more difficult than I thought.  I don't know which instantiation of SPI to set the clock or baudrate.  Is this exposed through the Adafruit_ILI9341 interface?  How would I know what the handle is?  Could you provide a minimum example that might do this?  tft is the module that calls SPI, not me.  I did try tft.begin(48000000);  That doesn't do anything, nor does 48000001.  Still 24MHz.  Do I make a #define in Adafruit_ILI9341.cpp for __SAMD51__  for SPI_DEFAULT_FREQ?

Or is this deep in the SPI library that I have to make this change?  Just confused about where the change goes, and how to get the correct instance.  Don't want to have to permanently maintain the spi lib for a single project, if I don't have to.  

Any additional ideas or leads would be helpful.  Thanks.  

begin, sends a command initSPI, but it only sends frequency, not the clock source.  begin also has access to setSPISpeed via Adafruit_SPITFT.h.  But there doesn't seem to be a way to pass this through to SPI itself.  SPI seems to over-ride the setting.  

I have tried define MAX_SPI 48000000 prior to including SPI.h and Adafruit_ILI9341.h.  But I don't see a way (my lack of knowledge) to setClockSource for the tft instance of SPI.  The version of SPI.cpp is found in .arduino15/packages/adafruit/hardware/samd/1.7.10/libraries/SPI

bdlabitt, 2022-05-08:
Also tried a 
`#define MAX_SPI 48000000
#define SPICLOCK SERCOM_CLOCK_SOURCE_FCPU`
prior to including SPI.h and Adafruit_GFX.h and Adafruit_ILI9341.h and Adafruit_FT6206.h in my header file.
Still 24MHz.

bdlabitt, 2022-05-08:
For what it is worth, the PJRC lib ILI9341_t3n supports a 30MHz write rate and a 20MHz read.  That would be pretty nice if something like that worked on a Feather M4.  Unfortunately the PJRC lib is not compatible with Adafruit's.  The incompatibility seems to be in the dma section, or that's where the compiler starts complaining!

fvzeppelin, 2022-09-05:
Maybe related to [](https://github.com/arduino/ArduinoCore-samd/issues/680)?

mikeysklar, 2026-11-07:
Thanks for the report.
`begin(freq)` does reach the SPI hardware, but the chip rounds down to what its clock can divide to.
On a Feather RP2040 with an HX8357, from `spi_get_baudrate()`:

| begin(freq) | actual clock |
|---|---|
| 8 MHz | 8 MHz |
| 16 MHz | 12 MHz |
| 32 MHz | 24 MHz |
| 64 MHz | 24 MHz |

[image]

SAMD51 works the same way from a 48 MHz reference, so 32 MHz gives 24 MHz.
To go faster there, call `SPI.setClockSource(SERCOM_CLOCK_SOURCE_100M)` before `begin()`, as dhalbert noted above.
Closing this one.

closed 2026-11-07 by mikeysklar