adafruit/Adafruit-GFX-Library #276: Lack of reference position for characters and changing to baseline from bottom-line when changing font

HamidSaffari, 2020-03-13:
Hi.
Would you implement set font  reference position in your library like what Ucglib has ucg.setFontPosTop(); 
Because when using setfont() it changes the top left to baseline.
Or at least don't change reference when changing fonts.

HamidSaffari, 2020-03-13:
I did that by adding :
	if(_setFontPosTop){
		x = x-xo;
		y = y-yo;
	}
inside Adafruit_GFX::drawChar() in line 1197.
and edit:
    setFont(const GFXfont *f = NULL , bool setFontPosTop =true);
void Adafruit_GFX::setFont(const GFXfont *f, bool setFontPosTop) {
......
....
  _setFontPosTop = setFontPosTop;
}
bool _setFontPosTop;
//=====================================================
and it works but the lower case characters stuck to top line and the result is a bit ugly.

mikeysklar, 2026-11-10:
This can be done with the current API, no library change needed. `getTextBounds()` at (0, 0) returns `y1`, the distance from the baseline up to the top of the text (a negative number), so:

```cpp
int16_t x1, y1;
uint16_t w, h;
tft.getTextBounds("H", 0, 0, &x1, &y1, &w, &h);
tft.setCursor(x, top - y1); // top of the text sits on "top"
tft.print("Hello xyz gjpq");
```

Measuring the string itself puts its own tallest glyph on the line, so `xyz` rides higher than `Hello` (white rows). Measuring a reference like `"H"` once keeps every string on one baseline (cyan rows), which avoids the lowercase-stuck-to-the-top problem from the per-glyph patch above. Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing, red line is `top`:

[image]

Closing since the existing API covers it, please reopen if there is a case this doesn't handle.

closed 2026-11-10 by mikeysklar