adafruit/Adafruit-GFX-Library #277: Arduino Due Hardware SPI support (or not working if supported)

JakeQuinn77, 2020-03-24:
This is in relation to the [issue ](https://github.com/adafruit/Adafruit-SSD1351-library/issues/28) of a library that inherited from this library SSD1315

If you may have noticed from the linked issue, im having trouble making Hardware SPI working it would seem that that library does not handle the SPI but is instead passed to this library.

on [line 54 of  the SPITFT.h](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.h)

`// Arduino Due?
typedef uint32_t ADAGFX_PORT_t; ///< PORT values are 32-bit
// USE_FAST_PINIO not available here (yet)...Due has a totally different
// GPIO register set and will require some changes elsewhere (e.g. in
// constructors especially).
#endif
#else                                      // !ARM
// Probably ESP8266 or ESP32. USE_FAST_PINIO is not available here (yet)
// but don't worry about it too much...the digitalWrite() implementation
// on these platforms is reasonably efficient and already RAM-resident,
// only gotcha then is no parallel connection support for now.
typedef uint32_t ADAGFX_PORT_t; ///< PORT values are 32-bit
#endif                                     // end !ARM`

_will require some changes elsewhere (e.g. in constructors especially)._ with that line i am not sure if the DUE is even supported. Which constructors need changing so i can change them

I asked everywhere for help, and it looks like this is the last place. 

- Arduino board:  DUE

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.12

- [Example TEST sketch of SSD1351 library](https://github.com/adafruit/Adafruit-SSD1351-library/blob/master/examples/test/test.ino)

ladyada, 2020-03-24:
hi sure, please submit a PR if you have a fix. we have no ETA on when we would look at this fix, we don't recommend the Due in general - its not very well supported

JakeQuinn77, 2020-03-24:
> hi sure, please submit a PR if you have a fix. we have no ETA on when we would look at this fix, we don't recommend the Due in general - its not very well supported

Thats unfortunate the DUE is arduinos fastest board perfect for a display (exempt the portenta that thing basically used the fastest microcontroller available).  I will try what i can do with the limit resources i have. If i ever get it sorted out I will suggest the necesarry edits

ladyada, 2020-03-24:
you could try metro m4
https://www.digikey.com/product-detail/en/adafruit-industries-llc/3382/1528-2605-ND/9342520
it is 120MHz
https://learn.adafruit.com/adafruit-metro-m4-express-featuring-atsamd51

JakeQuinn77, 2020-03-24:
I have found the problem, and its not with this library it is with the SSD1351 library. i have posted a possible fix in the [issue thread](https://github.com/adafruit/Adafruit-SSD1351-library/issues/28) there since this library there is no need to edit this library for my suggested fix.

ladyada, 2020-04-02:
great!

closed 2020-04-02 by ladyada