adafruit/Adafruit-GFX-Library #281: No solution yet to replace texts from external fonts?

LukeDaniel16, 2020-04-14:
I use the FreeSevenSegNumFont font and type values up to 4 digits in a square
tft.fillRoundRect (7, 20, 274, 138, 7, BLACK);
This text needs to be constantly replaced, about 0.7 seconds apart, using a 16x2 lcd monitor I can replace the text very quickly using the "print (" ") function, but when I try to rewrite that square, it it takes almost 1 second to be replaced, which causes a terrible visual effect.

My tft card is an [https://www.kuongshun-ks.com/uno/uno-board-shield/3-5-inch-ili9486-tft-shield-lcd-module.html](ili9481)

How can I draw a rectangle this size faster? 
- Arduino board: Atmega 2560

- Arduino IDE version (found in Arduino -> About Arduino menu):  Arduino IDE 1.8.12

pljakobs, 2020-04-14:
What you'll need to do is draw your new content on a canvas and then copy that to screen.
I have done this years ago with little uptake, it's probaby defunct with the newer versions, but if you look here, you might get an idea: https://github.com/pljakobs/Arduino_Gauges
Specifically look for textGauge::redraw(){
The whole business of mocking around with placement is probably something you can ignore

LukeDaniel16, 2020-04-14:
The problem is that depending on the size of the screen (I use an ILI9486, 3.5th) it can be said that there is a delay in the redesign, it is visibly ugly.

pljakobs, 2020-04-14:
@LukeDaniel16 yes, and when you print your text to a canvas object instead of the screen directly, the rewrite will not be character by character (which is what I assume yu find unpleasant) but the whole rect.

the idea is: 
1. print to canvas (it's a display object as any other as well)
2. copy the canvas to screen at the correct position
repeat

You might also try to see if my Gauges library still works with the current gfx library (I doubt it and it's really ugly code in parts) but if it does, you can just create a gauge, position it on screen and after that, all you have to do is update the gauge value.

PaintYourDragon, 2020-04-14:
Even an Arduino Mega lacks sufficient RAM to buffer a 1-bit canvas that size (which normally is the Official and Canonical Supported way of handling “clean” text overwrite with custom fonts).
In addition, AVRs are limited to 8 MHz SPI max and no DMA, and that screen has a LOT of pixels.
Really recommend moving to a 32-bit microcontroller; SAMD, ESP32, anything with the RAM and performance to handle fast refresh on a display of that size and depth. Closing this since it’s not a code issue but a resource issue.

closed 2020-04-14 by PaintYourDragon

pljakobs, 2020-04-15:
@PaintYourDragon since this is a recurring topic, would it make sense to implement something like a "text field" object - a simplified version of what I was trying to do?
Print to the text field and it copies to screen automatically. In a way, it's not really what a base gfx library should do, but then again, a base library wouldn't have buttons ;-)

PaintYourDragon, 2020-04-15:
Not really. GFX has already gotten pretty grotesque for something that’s just supposed to get a user from zero to Hello World. We just got LittlevGL working (see Adafruit_LvGL_Glue repo) and that’s way WAY more promising for Actual Proper UI Stuff, if one has a 32-bit MCU to throw at it.

LukeDaniel16, 2020-04-29:
Really, using esp32 or Stm microprocessors the speed of atualization is very better, and his still low cost.