adafruit/Adafruit-GFX-Library #283: Online Playground for the Adafruit GFX Library

urish, 2020-04-23:
I took the [ssd1306_128x64_i2c](https://github.com/adafruit/Adafruit_SSD1306/blob/master/examples/ssd1306_128x64_i2c/ssd1306_128x64_i2c.ino) example and turned it into an online playground on Wokwi, an Arduino teaching platform I'm working on. 

The playground allows you to run the code *in your browser*, using an [Arduino AVR simulator](https://blog.wokwi.com/avr8js-simulate-arduino-in-javascript/?utm_source=github&utm_medium=social&utm_campaign=Adafruit_GFX), so you can learn how the Adafruit GFX API works and experiment with it right from your browser, without the need to pull and connect an Arduino board to your PC:

[https://wokwi.com/playground/ssd1306](https://wokwi.com/playground/ssd1306?utm_source=github&utm_medium=social&utm_campaign=Adafruit_GFX)

I also created another example with the [Tamaguino](https://alojzjakob.github.io/Tamaguino/) game, which also uses the Adafruit GFX library:

[https://wokwi.com/playground/tamaguino](https://wokwi.com/playground/tamaguino?utm_source=github&utm_medium=social&utm_campaign=Adafruit_GFX)

I'd love to get your input on:

1. Does it work well for you?
2. Does it feel intuitive?
3. Which use cases come to your mind?
4. What else would you like to see?

Thanks!

ddlsmurf, 2021-04-24:
In my case, the dino has a display issue (like it's shifted right by half a screen) [image] - but the first example works well.

Yes it's intuitive. It's exciting to me if you can support more screens, I would use this to play around with rendering code without flashing a device at each test. Thanks !

urish, 2021-04-25:
Thanks for the feedback @ddlsmurf!

First the [tamaguino](https://wokwi.com/arduino/projects/296836708191699465) example, does it work well for you now?

Which other screens would you like to see supported?

ddlsmurf, 2021-04-25:
Yup works fine now - the timer in the top right is very annoying though it really needs a fixed width font. The screen I'm playing with now is 200x200 but ideally it could be cool to just pick any resolution

urish, 2021-04-25:
Regarding the timer - the font should be fixed with. This is what it looks here (Chrome on Windows):

https://user-images.githubusercontent.com/892318/116008818-29c67100-a61f-11eb-8c5c-cb3024199571.mp4

Does it look different for you?

What model is that screen?

urish, 2021-04-25:
Update: I managed to reproduce the jumpy timer issue on Safari and uploaded a fix. Does it work better for you now?

ddlsmurf, 2021-04-25:
Perfect =) Impressive work alltogether.

lemonkey, 2021-08-04:
@urish Thanks a lot for this. Dramatically sped up some of my development work.

urish, 2021-08-04:
Yes, they should work out of the box. e.g. https://wokwi.com/arduino/projects/305978006798598721

mikeysklar, 2026-10-21:
Thanks for sharing the Wokwi playground, @urish.
It is a great way to try GFX without hardware.
Closing this one since it is a showcase, not an issue.

closed 2026-10-21 by mikeysklar