adafruit/Adafruit-GFX-Library #288: SSD1315 skipping lines

caternuson, 2020-05-14:
Several cases:

- https://forums.adafruit.com/viewtopic.php?f=47&t=165405
- https://forums.adafruit.com/viewtopic.php?f=47&t=163320
- https://forums.adafruit.com/viewtopic.php?f=19&t=162264

with SSD1351 based 1.5" color OLED: https://www.adafruit.com/product/1431

makermelissa, 2020-06-08:
Something I noticed looking at those posts, it looks like this is specific to 8-bit processors (drawing code varies by proc type sometimes).

makermelissa, 2020-06-08:
I have been unable to reproduce this yet. I have tried an ESP32, Metro 328, and even an old Arduino Diecimila (I don't own a genuine uno or nano). I've tried the graphics test using both hardware and software SPI and the imagereader examples and those all worked perfectly. So there must be something I'm missing here.

CCDengineer, 2020-06-09:
I have this issue with a new genuine UNO Rev3.  The problem seems to have been introduced between versions 1.1.1 and 1.2.0 of the  ssd1351 library.  The test program works fine if I revert back to the 1.1.1 version.

nygma2004, 2020-06-11:
I received my SSD1351 1.5 recently, and I am having the same issue. I have the 1.2.4 version installed. I am testing it on a Wemos D1 mini with the following pin assignments:
#define SCLK_PIN 14
#define MOSI_PIN 13
#define DC_PIN   5
#define CS_PIN   4
#define RST_PIN  3

makermelissa, 2020-06-11:
Thanks, I have a D1 mini.

makermelissa, 2020-06-16:
Tested with D1 Mini and it still is not skipping lines. This is an odd one that seems to affect only certain displays. Unless there were a couple different versions of the display or something, I'm not sure what the difference would be. I mean I guess it's possible somebody's libraries or Board Support Package may not be completely up to date. Another possibility is perhaps there's an older set of libraries overriding these such as the outdated ones that were being included with the Teensy software (I'm not sure if they still are).

nygma2004, 2020-06-17:
Hi Melissa,
I did not know that these can have such an impact on the compiled code. So I have updated everything. I downloaded the latest version of IDE which is 1.8.13 at the moment. I have gone into the board manager and updated all the board manager definitions and also updated all my libraries (even though the Adafruit was already up-to-date). I recompiled the code and it is working now.

Just for reference to anyone else, I am using this [1.5 inch 128x128 OLED screen from ICStation.com](http://www.icstation.com/15inch-128x128-oled-display-screen-module-ssd1351-driver-128128-power-p-14400.html). 
And for anyone who stumbles upon this issue, this is the wiring I used, and these are the only lines that I have modified in the example sketch:
```
// Wiring on a Wemos D1 Mini (ESP8266)
// GND -> GND
// Vcc -> 3.3v
#define SCLK_PIN 14   // SCL -> D5
#define MOSI_PIN 13   // SCA -> D7
#define DC_PIN   5    // DC  -> D1
#define CS_PIN   4    // CS  -> D2
#define RST_PIN  3    // RES -> RX 
```

LittleSpark55, 2020-06-19:
Hello good people, I am somewhat a newbie here... making a solar-powered bike computer here for a trip around  the Okefenokee swamp.    https://en.wikipedia.org/wiki/Okefenokee_Swamp

I've copied and slightly modified some code and got it to work on small .96 OLED using the ssd1306 library. Then got a larger 1.5in (128x128), ssd1351 OLED.   My screen needs to be refreshed and redrawn (code line 124) but it fails and gives me an "'class Adafruit_SSD1351' has no member named 'display' " and display.clear(); also errors out.
How do I clear without flickering?  I read that "The Adafruit library doesn't clear areas that have not changed,  it rewrites these areas with the same data." 

// This program is for drawing a chart and moving dots (illusion of moving wave) to display voltages or current levels
// Arduino UNO and SSD1351 128X128 OLED display
// 6/17/2020

// Screen dimensions
#define SCREEN_WIDTH  128
#define SCREEN_HEIGHT 128 

// U8GLIB_SSD1351_128X128_332 u8g(   13,        11,      8,      9,         7); 
// Arduino UNO: SW SPI Com:    SCK = 13, MOSI = 11, CS = 8, A0 = 9, RESET = 7 

// You can use any (4 or) 5 pins 
#define SCLK_PIN 13
#define MOSI_PIN 11
#define DC_PIN   9
#define CS_PIN   8
#define RST_PIN  7

// Color definitions
#define BLACK           0x0000
#define BLUE            0x001F
#define RED             0xF800
#define GREEN           0x07E0
#define CYAN            0x07FF
#define MAGENTA         0xF81F
#define YELLOW          0xFFE0  
#define WHITE           0xFFFF

#include <SPI.h>
#include <Wire.h>
#include "SPI.h"                    // YT Includes library for SPI communication of display
#include <Adafruit_GFX.h>           //Includes core graphics library
#include <Adafruit_SSD1351.h>       //Includes hardware specific library

 
Adafruit_SSD1351 display = Adafruit_SSD1351(SCREEN_WIDTH, SCREEN_HEIGHT, &SPI, CS_PIN, DC_PIN, RST_PIN);

byte count;
byte sensorArray[128];
byte drawHeight;

char filled = 'D'; //decide either filled or dot display (F or D = dot, any else filled)
char drawDirection = 'R'; //decide drawing direction, from right or from left (L=from left to right, any else from right to left)
char slope = 'W'; //slope colour of filled mode white or black slope (W=white, any else black. Well, white is blue in this dispay but you get the point)

      
void setup(void) 
  {
       Serial.begin(9600);
       //Serial.print("InSETUP");
       display.begin();
       display.fillScreen(0x0000);  // This clears entire screen (BLACK)
  }

void loop() {      
  //This section puts moving dots on chart (voltage or current levels)  
  drawHeight = map(analogRead(A0), 0, 1023, 4, 64 );  //  0Min/1023Max (on chart), 4Min/64Max (limits of moving dots) 
  sensorArray[0] = drawHeight;                        //vertical limit of moving dots

  for (count = 1; count <= 80; count++ )              //80  Position of moving chart start
  {
    if (filled == 'D' || filled == 'd')
    {
      if (drawDirection == 'L' || drawDirection == 'l')
      {
[... 79 lines cut ...]
  
  display.print(drawHeight-4);  // Numerical value
  
  display.setCursor(90, 8);
  display.setTextSize(1);
  display.setTextColor(WHITE);  // BLACK,BLUE,RED,GREEN,CYAN,MAGENTA,YELLOW,WHITE
  display.print("KM/h");        // Label

  display.drawLine( 0, 0,  0, 60, WHITE);   //  64//32 );  // Left  Vert line on graph Horiz/Vertical
  display.drawLine(80, 0, 80, 60, WHITE);   //  64//32 );  // Right Vert line on graph

  for (count = 0; count < 70; count += 10)   //40-> 60  //Number of horiz marks
      {
    display.drawLine(80, count, 75, count, WHITE);   //Length / Position of RIGHT horiz marks,
    display.drawLine( 5, count,  0, count, WHITE);   //Length / Position of LEFT  horiz marks,
      }

  for (count = 10; count < 80; count += 10)   //80  Num of Horiz Pixels 
      {
    display.drawPixel(count,  0 , WHITE);   // Horizontal dots on graph
    display.drawPixel(count, 30 , WHITE);   //30
    display.drawPixel(count, 60 , WHITE);   //60
      }
  
}

makermelissa, 2020-07-27:
@nygma2004, for issues with updating your code, the forums would probably be better suited to your needs and doesn't appear to be related to this issue.

As for the failing lines of code, I don't think you need to call the display() function, nor does it exist and maybe try calling `display.fillScreen(BLACK);` to clear the screen. I think the main difference is that the SSD1306 probably uses a FrameBuffer due to being small and monochrome whereas this one doesn't. I'd suggest looking at the included examples for how it's done for this display.

makermelissa, 2020-07-27:
@LittleSpark55, for issues with updating your code, the forums would probably be better suited to your needs and doesn't appear to be related to this issue.

As for the failing lines of code, I don't think you need to call the display() function, nor does it exist and maybe try calling display.fillScreen(BLACK); to clear the screen. I think the main difference is that the SSD1306 probably uses a FrameBuffer due to being small and monochrome whereas this one doesn't. I'd suggest looking at the included examples for how it's done for this display.

makermelissa, 2020-07-27:
I'm going to close this issue because it seems to be related to using outdated dependencies and I have been unable to reproduce otherwise. With updated libraries it solves the issue, so there's not really anything we can do on the library itself. It's possible that there are some outdated versions of libraries installed with other packages such as the Teensy and these should probably be evaluated on a case-by-case basis.

snail23, 2023-02-16:
This issue still exists in 2023 using all of the latest library versions. Board is a Daisy Seed, STM32H7 MCU, and WaveShare SSD1351 display. I draw bitmaps as well as text on my device and this issue is happening with both.

makermelissa, 2023-02-16:
Hi @snail23, I can open this again for now. Just to see if the issue has reappeared, I can test it out on one of our SSD1351 displays and maybe something like the Adafruit Feather STM32F405 Express. However, if I'm not seeing the lines issue, it may either be related to the STM32H7 or WaveShare's display, of which I have neither nor can officially support.

I'm working on wrapping up some other projects and I don't have easy access to either board at the moment due to limited mobility, so it may be a bit before I can actually get around to testing.

mikeysklar, 2026-11-14:
Tested a 1.5" SSD1351 on a Feather M0 Express with hardware SPI at four clock speeds.

The test streams diagonal stripes row by row, so a dropped row would break the diagonals.

| SPI clock | Full-screen draw | Result |
|---|---|---|
| 4 MHz | 124 ms | clean |
| 8 MHz (default) | 90 ms | clean |
| 12 MHz | 80 ms | clean |
| 24 MHz | 71 ms | panel ignores the frame and keeps the old image |

[image]

No skipped lines up to 12 MHz.

Too fast a clock makes the panel drop data, so on a fast chip like the STM32H7 it is worth checking what SPI clock you really get.

@snail23 does `tft.begin(4000000)` fix it on your Daisy Seed?

<details>
<summary>Test sketch</summary>

```cpp
// #288: look for skipped lines on a 1.5" SSD1351 at several SPI clocks.
// Feather M0 Express, hardware SPI, CS 9, DC 10, RST 11.
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1351.h>
#include <SPI.h>

#ifndef SPI_HZ
#define SPI_HZ 8000000
#endif

Adafruit_SSD1351 tft(128, 128, &SPI, 9, 10, 11);
uint16_t row[128];

void setup() {
  Serial.begin(115200);
  tft.begin(SPI_HZ);
  tft.setRotation(3);
  tft.fillScreen(0);

  uint32_t t = micros();
  tft.startWrite();
  tft.setAddrWindow(0, 0, 128, 128);
  for (int y = 0; y < 128; y++) {
    for (int x = 0; x < 128; x++)
      row[x] = ((x + y) / 8) & 1 ? 0xFFFF : 0x001F;
    tft.writePixels(row, 128);
  }
  tft.endWrite();
  t = micros() - t;

  for (int y = 52; y < 76; y += 2)
    tft.drawFastHLine(16, y, 96, 0xF800);
  tft.fillRect(30, 56, 68, 16, 0);
  tft.setTextColor(0x07E0);
  tft.setCursor(34, 60);
  tft.print(SPI_HZ / 1000000);
  tft.print(" MHz");

  for (int i = 0; i < 30 && !Serial; i++)
    delay(100);
  Serial.print("#288 SPI_HZ=");
  Serial.print(SPI_HZ);
  Serial.print(" fill_us=");
  Serial.println(t);
}

void loop() {}
```

</details>

mikeysklar, 2026-11-16:
@snail23 I tried the same test on a Feather RP2040, which can run SPI much faster than the M0.

| Requested SPI clock | Result |
|---|---|
| 12, 16, 20 MHz | Clean, no skipped lines |
| 24, 32, 48, 62.5 MHz | Panel ignores the data and keeps the old picture |

[image]

Left to right: 16, 20 and 24 MHz.

The 24 MHz run still shows "20 MHz" from the run before it.

The SSD1351 needs at least 50 ns per SPI clock, so about 20 MHz is the limit.

Adafruit's CircuitPython SSD1351 driver also caps SPI at 16 MHz.

Near that limit, long wires can drop some lines instead of all of them.

Try `tft.begin(8000000)` on the Daisy Seed.

Closing for now, since this looks like SPI speed or wiring and not GFX.

Please comment here if the lines are still there at 8 MHz, and we will reopen it.

<details>
<summary>Test sketch</summary>

```cpp
// #288: look for skipped lines on a 1.5" SSD1351 at several SPI clocks.
// Feather RP2040 (or M0 Express), hardware SPI, CS 9, DC 10, RST 11.
// Diagonal stripes are streamed row by row through one address window,
// so a dropped row shifts everything below it and breaks the diagonals.
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1351.h>
#include <SPI.h>

#ifndef SPI_HZ
#define SPI_HZ 8000000
#endif

Adafruit_SSD1351 tft(128, 128, &SPI, 9, 10, 11);
uint16_t row[128];

void setup() {
  Serial.begin(115200);
  tft.begin(SPI_HZ);
  tft.setRotation(3);
  tft.fillScreen(0);

  uint32_t t = micros();
  tft.startWrite();
  tft.setAddrWindow(0, 0, 128, 128);
  for (int y = 0; y < 128; y++) {
    for (int x = 0; x < 128; x++)
      row[x] = ((x + y) / 8) & 1 ? 0xFFFF : 0x001F;
    tft.writePixels(row, 128);
  }
  tft.endWrite();
  t = micros() - t;

  // 1-pixel horizontal lines every other row in the middle band.
  for (int y = 52; y < 76; y += 2)
    tft.drawFastHLine(16, y, 96, 0xF800);
  tft.fillRect(30, 56, 68, 16, 0);
  tft.setTextColor(0x07E0);
  tft.setCursor(34, 60);
  tft.print(SPI_HZ / 1000000);
  tft.print(" MHz");

  for (int i = 0; i < 30 && !Serial; i++)
    delay(100);
  Serial.print("#288 SPI_HZ=");
  Serial.print(SPI_HZ);
  Serial.print(" fill_us=");
  Serial.println(t);
}

void loop() {}
```

</details>

closed 2026-11-16 by mikeysklar