adafruit/Adafruit-GFX-Library #291: Complier errors beginning with release 1.8.0 to 1.8.2

Pconti31, 2020-05-18:
"C:\\arduino-1.8.12\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\arduino-1.8.12\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\arduino-1.8.12\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Paul\\Documents\\Arduino\\libraries\\GUIslice\\src" "-IC:\\Users\\Paul\\Documents\\Arduino\\libraries\\Adafruit-GFX-Library" "-IC:\\Users\\Paul\\Documents\\Arduino\\libraries\\Adafruit_ILI9341" "-IC:\\arduino-1.8.12\\hardware\\arduino\\avr\\libraries\\SPI\\src" "-IC:\\arduino-1.8.12\\hardware\\arduino\\avr\\libraries\\Wire\\src" "-IC:\\Users\\Paul\\Documents\\Arduino\\libraries\\Adafruit_STMPE610" "C:\\Users\\Paul\\Documents\\Arduino\\libraries\\Adafruit-GFX-Library\\Adafruit_MonoOLED.cpp" -o nul
Alternatives for Adafruit_I2CDevice.h: []
ResolveLibrary(Adafruit_I2CDevice.h)
  -> candidates: []
In file included from C:\Users\Paul\Documents\Arduino\libraries\Adafruit-GFX-Library\Adafruit_MonoOLED.cpp:18:0:

C:\Users\Paul\Documents\Arduino\libraries\Adafruit-GFX-Library\Adafruit_MonoOLED.h:28:10: fatal error: Adafruit_I2CDevice.h: No such file or directory

 #include <Adafruit_I2CDevice.h>

          ^~~~~~~~~~~~~~~~~~~~~~

compilation terminated.

Using library GUIslice at version 0.14.0 in folder: C:\Users\Paul\Documents\Arduino\libraries\GUIslice 
Using library Adafruit-GFX-Library at version 1.8.0 in folder: C:\Users\Paul\Documents\Arduino\libraries\Adafruit-GFX-Library 
Using library Adafruit_ILI9341 at version 1.5.4 in folder: C:\Users\Paul\Documents\Arduino\libraries\Adafruit_ILI9341 
Using library SPI at version 1.0 in folder: C:\arduino-1.8.12\hardware\arduino\avr\libraries\SPI 
Using library Wire at version 1.0 in folder: C:\arduino-1.8.12\hardware\arduino\avr\libraries\Wire 
Using library Adafruit_STMPE610 at version 1.1.0 in folder: C:\Users\Paul\Documents\Arduino\libraries\Adafruit_STMPE610 
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

PaintYourDragon, 2020-05-18:
Adafruit_BusIO is now a dependency. Can install from Library manager or at https://github.com/adafruit/Adafruit_BusIO

closed 2020-05-18 by PaintYourDragon

tushada, 2021-03-21:
After trying to resolve the Compilation issue from 3 hours - I found your Your Solution of adding the library for Adafruit_BusIO - it resolved the error. Thanks a lot - you have given the right solution.

planktonfun, 2021-04-06:
Weird I kept installing Adafruit_BusIO but Im still stuck with
Arduino\libraries\Adafruit_GFX_Library\Adafruit_GrayOLED.h:30:32: fatal error: Adafruit_I2CDevice.h: No such file or directory
 #include <Adafruit_I2CDevice.h>

alhubbar673, 2021-05-28:
Adding the Adafruit_BusIO to my libraries worked for me!