adafruit/Adafruit-GFX-Library #298: Use adafruit with platformio under ststm32, can not complie, too many errors?

KiraVerSace, 2020-05-28:
OK, I use the library in vscode with platformIO, I just install the library adafruit_gfx, and adafruit_st7789, and some libraries depend on it, it install automatically.
After it, I start to use it like this:
```
#include <Arduino.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ST7789.h>
#include <SPI.h>

SPIClass TFT_SPI(PB5, PB4, PB3);

Adafruit_ST7789 tft = Adafruit_ST7789(&TFT_SPI, -1, PB6, -1);

void setup() 
{
	Serial.begin(115200);
	pinMode(LED_GREEN, OUTPUT);
	Serial.println("Hello");
	SerialLP1.begin(115200);

}
```
I just build it, there are too many errors as blew. Would you like to help me, thank you very much.
```
> Executing task: platformio run <

Processing nucleo_l476rg (platform: ststm32; board: nucleo_l476rg; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_l476rg.html
PLATFORM: ST STM32 6.1.0 > ST Nucleo L476RG
HARDWARE: STM32L476RGT6 80MHz, 128KB RAM, 1MB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-arduinoststm32 4.10800.200207 (1.8.0) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 25 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit GFX Library> 1.8.3
|   |-- <Adafruit BusIO> 1.3.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit ST7735 and ST7789 Library> 1.5.15
|   |-- <Adafruit GFX Library> 1.8.3
|   |   |-- <Adafruit BusIO> 1.3.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <SPI> 1.0
|   |-- <Adafruit seesaw Library> 1.3.1
|   |-- <SD>
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio/build/nucleo_l476rg/lib90b/Adafruit BusIO_ID6214/Adafruit_BusIO_Register.cpp.o
Compiling .pio/build/nucleo_l476rg/lib90b/Adafruit BusIO_ID6214/Adafruit_I2CDevice.cpp.o
In file included from /Users/kiraversace/.platformio/lib/Adafruit BusIO_ID6214/Adafruit_BusIO_Register.h:1,
                 from /Users/kiraversace/.platformio/lib/Adafruit BusIO_ID6214/Adafruit_BusIO_Register.cpp:1:
/Users/kiraversace/.platformio/lib/Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1:10: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

    1 | #include <Wire.h>
[... 173 lines cut ...]
      |      ^~~~~~~~
/Users/kiraversace/.platformio/lib/SD_ID161/utility/Sd2Card.cpp:91:8: error: 'clkport' was not declared in this scope
   91 |       *clkport |=  clkpinmask;
      |        ^~~~~~~
/Users/kiraversace/.platformio/lib/SD_ID161/utility/Sd2Card.cpp:95:13: error: 'misoport' was not declared in this scope
   95 |       if ((*misoport) & misopinmask)  data |= 1;
      |             ^~~~~~~~
/Users/kiraversace/.platformio/lib/SD_ID161/utility/Sd2Card.cpp: In member function 'uint8_t Sd2Card::init(uint8_t, uint8_t, int8_t, int8_t, int8_t)':
/Users/kiraversace/.platformio/lib/SD_ID161/utility/Sd2Card.cpp:302:5: error: 'clkport' was not declared in this scope
  302 |     clkport     = portOutputRegister(digitalPinToPort(clockPin_));
      |     ^~~~~~~
/Users/kiraversace/.platformio/lib/SD_ID161/utility/Sd2Card.cpp:304:5: error: 'mosiport' was not declared in this scope
  304 |     mosiport    = portOutputRegister(digitalPinToPort(mosiPin_));
      |     ^~~~~~~~
/Users/kiraversace/.platformio/lib/SD_ID161/utility/Sd2Card.cpp:306:5: error: 'misoport' was not declared in this scope
  306 |     misoport    = portInputRegister(digitalPinToPort(misoPin_));
      |     ^~~~~~~~
/Users/kiraversace/.platformio/lib/SD_ID161/utility/Sd2Card.cpp: In member function 'uint8_t Sd2Card::readData(uint32_t, uint16_t, uint16_t, uint8_t*)':
/Users/kiraversace/.platformio/lib/SD_ID161/utility/Sd2Card.cpp:438:12: warning: unused variable 'n' [-Wunused-variable]
  438 |   uint16_t n;
      |            ^
*** [.pio/build/nucleo_l476rg/lib54b/SD_ID161/utility/Sd2Card.cpp.o] Error 1
===================================================================================== [FAILED] Took 2.99 seconds =====================================================================================
终端进程已终止，退出代码: 1
```

ladyada, 2020-05-28:
hihi yes this is an issue with platformio's greedy compilation strategy - it compiles files even when they are not used by the project
@ivankravets can help you configure the libraries to not compile all files

closed 2020-05-28 by ladyada

ivankravets, 2020-05-28:
@KiraVerSace please ignore not needed libraries with https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

KiraVerSace, 2020-05-28:
> hihi yes this is an issue with platformio's greedy compilation strategy - it compiles files even when they are not used by the project
> @ivankravets can help you configure the libraries to not compile all files

> @KiraVerSace please ignore not needed libraries with https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

Hi, Two great gods I am not bright enough and from China, Could you help me more detailed,
I just want to use my IPS_TFT with ST7789 under the stm32duino and use the SPI1,
which library should I ignored and where should I changed in files.

But I never meet the problem berfore I update the library in the platformIO,

Thankyou very much.
Best regards!

valeros, 2020-05-28:
Hi @KiraVerSace ! Your `platformio.ini` should look something like this:
```
[nucleo_l476rg]
platform = ststm32
board = nucleo_l476rg
framework = arduino
lib_deps = 
    Adafruit GFX Library
    Adafruit ST7735 and ST7789 Library
    Wire
lib_ignore = 
    SD
```

KiraVerSace, 2020-05-29:
> @KiraVerSace please ignore not needed libraries with https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

> Hi @KiraVerSace ! Your `platformio.ini` should look something like this:
> 
> ```
> [env:nucleo_l476rg]
> platform = ststm32
> board = nucleo_l476rg
> framework = arduino
> lib_deps = 
>     Adafruit GFX Library
>     Adafruit ST7735 and ST7789 Library
>     Wire
> lib_ignore = 
>     SD
> ```

Thak you for your help, It seems slove my problem, I just add lib_ignore = SD,
but I want to know why, because I used to use the adafruit library and I don't meet this problem, But now, so I want to know, what's the reason cause it?
and I never use lib_deps before.

ladyada, 2020-05-29:
it happens now because we now document all dependancies for libraries, as the Arduino IDE now auto-installs them - so there are more dependancies that are compiled