adafruit/Adafruit-GFX-Library #299: How to use fonts like freesans5pt7b.h file in stm

25prithvi, 2020-05-28:
So I want to use the fonts .h files in STM which have .c files, but not .cpp file.

I went through the .h files of different fonts and where there are 2 structures i.e GFXfont and GFXglyph.

So how to decode it to get a character that I can print on the screen?

If you can provide a flowchart or algorithm for decoding the font .h file then it'll really help me a lot.
Basically how to use the bitmap, bitmapoffset, etc.

I am able to print different logos using the bitmap but I am not able to use the bitmap in these fonts .h file.

BillyDonahue, 2020-05-28:
This guide might help. It features a deep dive into the font format.

https://learn.adafruit.com/creating-custom-symbol-font-for-adafruit-gfx-library/understanding-the-font-specification

mikeysklar, 2026-10-21:
This Learn guide explains the GFXfont and GFXglyph format in detail.

https://learn.adafruit.com/creating-custom-symbol-font-for-adafruit-gfx-library/understanding-the-font-specification

drawChar() in Adafruit_GFX.cpp is a working reference decoder for these fonts.

Closing this one.

closed 2026-10-21 by mikeysklar