adafruit/Adafruit-GFX-Library #304: drawRGBBitmap on SSD1351

nygma2004, 2020-06-17:
I have used drawRGBBitmap on a neopixel display in the past and I wanted to reuse the same logic on an SSD1351 128x128 display. I am using the drawRGBBitmap because I would like to display a full screen animation that I store as frames in PROGMEM.
I created my variable as: 
`static const uint16_t PROGMEM bitmap_frames[][16384] = {....};`
since 128x128 is 16384 pixels. I used LCD Image Converted to convert my image from bitmap to R5G6B5 format.

I am displaying the first frame as:
`tft.drawRGBBitmap(0,0,bitmap_frames[0],128,128);`

The image is getting displayed correctly, but 1/4 size on the top left corner and duplicated next to it to the right. So the image is showing twice on the top half of the screen. Colour and image aspect is correct.
All the examples I have found for displaying image is loading the images from the SD card and use drawBitmap, but not drawRGBBitmap. Is this not working for SSD1351?

nygma2004, 2020-07-05:
Is drawRGBBitmap is not supported on the SSD1351?

tsk15535904190, 2020-11-14:
我也遇到了相同的问题，请问你解决了吗？

nygma2004, 2020-11-14:
> 我也遇到了相同的问题，请问你解决了吗？

No, still waiting for a response from the author.

tsk15535904190, 2020-11-14:
I use stm32 ，succeed display image！

mikeysklar, 2026-11-13:
`drawRGBBitmap()` works with any display, the SSD1351 included.

A quarter-size image shown twice across the top half usually means the converter packed two pixels into each 32-bit word.

In a `uint16_t` array only one pixel of each pair survives, so each screen row gets two half-width image rows.

In LCD Image Converter, set the block size to 16 bit for R5G6B5.

Same 128x128 image both ways on a Feather HUZZAH ESP8266 + 3.5" HX8357 FeatherWing:

[image]

Closing, please reopen if 16-bit blocks still show it.

closed 2026-11-13 by mikeysklar