adafruit/Adafruit-GFX-Library #307: Platform IO compile error on every new project ( Wire.h / SPI.h )

xsrf, 2020-06-20:
Hi,
whenever I start a new project in Platform IO using Adafruit GFX Library I get a bunch of compile errors.

main.cpp
```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

platformio.ini
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = Adafruit GFX Library
```

<details> 
<summary>compile output</summary>

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.12.1 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing id=13
Using cache: C:\Users\***\.platformio\.cache\cb\eb4a83205a6bf7664c071bda082bc1cb
Adafruit GFX Library @ 1.9.0 has been successfully installed!
Installing dependencies
Looking for Adafruit BusIO library in registry
Found: https://platformio.org/lib/show/6214/Adafruit BusIO
LibraryManager: Installing id=6214
Using cache: C:\Users\***\.platformio\.cache\87\51e35a894cbff55174e7997f44b7d387
Adafruit BusIO @ 1.3.2 has been successfully installed!
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit GFX Library> 1.9.0
|   |-- <Adafruit BusIO> 1.3.2
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
Generating partitions .pio\build\esp32doit-devkit-v1\partitions.bin
Compiling .pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o
In file included from .pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214/Adafruit_BusIO_Register.h:1:0,
                 from .pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp:1:
.pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

compilation terminated.
In file included from .pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp:1:0:
.pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

compilation terminated.
In file included from .pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp:1:0:
.pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

compilation terminated.
Compiling .pio\build\esp32doit-devkit-v1\lib3d0\Adafruit GFX Library_ID13\Adafruit_GFX.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib3d0\Adafruit GFX Library_ID13\Adafruit_MonoOLED.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib3d0\Adafruit GFX Library_ID13\Adafruit_SPITFT.cpp.o
*** [.pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_I2CDevice.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_BusIO_Register.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib1a3\Adafruit BusIO_ID6214\Adafruit_SPIDevice.cpp.o] Error 1
In file included from .pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_MonoOLED.h:30:0,
                 from .pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_MonoOLED.cpp:20:
.pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214/Adafruit_I2CDevice.h:1:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib3d0\Adafruit GFX Library_ID13\Adafruit_MonoOLED.cpp.o] Error 1
In file included from .pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_SPITFT.cpp:36:0:
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_SPITFT.h:26:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib3d0\Adafruit GFX Library_ID13\Adafruit_SPITFT.cpp.o] Error 1
================================================================================================== [FAILED] Took 5.62 seconds ==================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```
</details>

I usually solve this by manually deleting 
```
.pio\libdeps\esp32doit-devkit-v1\Adafruit BusIO_ID6214
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_MonoOLED.cpp
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_MonoOLED.h
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_SPITFT.cpp
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_SPITFT.h
.pio\libdeps\esp32doit-devkit-v1\Adafruit GFX Library_ID13\Adafruit_SPITFT_Macros.h
```

I don't need them anyways.
I'm not sure what causes this. `Wire.h` / `SPI.h` clearly are in the include path. I use them in my projects all the time. Same with ESP8266 projects. Sometimes I just don't get compilers...

ladyada, 2020-06-20:
@ivankravets can help you with this issue - please open a ticket with platformio - Wire and SPI should come with any Arduino installation :)

xsrf, 2020-06-21:
Well, the thing is, `Wire.h` is included.

This compiles fine without Adafruit GFX:

```
#include <Arduino.h>
#include <Wire.h>

void setup() {
  Wire.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
```

Now, just adding `lib_deps = Adafruit GFX Library` to this code results in compile errors again.
Somehow Whire/SPI is not available to the Adafuit GFX library.
Some other libraries that do use Wire/SPI do not have this problem...

ladyada, 2020-06-21:
kk - have you opened a ticket with platformio?

xsrf, 2020-06-21:
Hmm, I don't have a deep understanding of how compilers create their include paths and actually why this fails. I'm not sure if it is an issue with Adafruit GFX, esp32/esp8266 for platformio, platformio in general or VScode... 
I don't want to spam lots of repos with this. I chose this one because I think it's a wide used library that is affected by this.

ladyada, 2020-06-21:
hi please open a ticket with platformio - this library works perfectly with arduino

xsrf, 2020-06-22:
got redirected to https://github.com/adafruit/Adafruit-GFX-Library/issues/298#issuecomment-636046011 🤷‍♂️

valeros, 2020-06-22:
Add the following lines to your code to help PlatformIO detect dependencies automatically:
```
#include <Wire.h>
#include <SPI.h>
...
```
or specify the dependencies explicitly in the `platformio.ini` file: 
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps =
    Adafruit GFX Library
    Wire
    SPI
```

xsrf, 2020-06-23:
Thx, that helps.

closed 2020-06-23 by xsrf

xsrf, 2021-11-21:
Btw. it now changed with recent versions. Now this is needed for the library to compile:
```ini
lib_deps = 
    adafruit/Adafruit GFX Library
    adafruit/Adafruit BusIO
```

aaronse, 2022-01-26:
Thanks to folks that doc'd this issue and fix.

**problem:** Build fail with Wire.h: No such file or directory
In file included from C:\users\aaron\.platformio\lib\Adafruit BusIO/Adafruit_BusIO_Register.h:9,
                 from C:\users\aaron\.platformio\lib\Adafruit BusIO\Adafruit_BusIO_Register.cpp:1:
C:\users\aaron\.platformio\lib\Adafruit BusIO/Adafruit_I2CDevice.h:5:10: fatal error: Wire.h: No such file or directory
**cause:** 🤷‍♂️ 
**fix:** Edit platform.io, add *Wire* to *lib_deps =* section
**env:** New project, added reference to Adafruit DotStar 1.2.0