adafruit/Adafruit-GFX-Library #308: How to detech what GFX Font format my sketch included...

KurtE, 2020-06-22:
Sorry I know I am not following all of the guidelines for issues.

The somewhat recent Pull Request: https://github.com/adafruit/Adafruit-GFX-Library/pull/286

which changed the font format, has broken several of our faster display drivers for the Teensy 3.x and 4.x which can use these fonts.   Including the ability to display the text in Opaque mode.  

We did  did not notice these changes as our older version of GFX that WAS included as part of Teensyduino still had the older format.  But the new release will most likely not include this older version of GFX.

The main change I noticed was the first and last character were changed from 8 bits to 16 bits, and as such I believe our code will punt from output any of these.  

Question is, is there a proper way to detect what version of code, as to adapt.  I have a couple of ideas including:

a) try to use your font header file.  Obviously the cleanest way...

b) Some way to detect which adafruit_gfx library the change was added and if I can detect this as part of the sketch?

c) Maybe big hack.  If last character is 0 probably new format, so grab the next byte and likewise patch up the field following it. 

Any other suggestions? 

Thanks
Kurt

ladyada, 2020-06-22:
oh oops you can just submit a PR to revert it, the person has decided to go with a fork of GFX anways

KurtE, 2020-06-22:
Thanks,  right now I think I have a way to work around it in a clean way... 
I added the __has_include part
```
// Lets see about supporting Adafruit fonts as well?
#if __has_include(<gfxfont.h>)
	#include <gfxfont.h>
#endif

#ifndef _GFXFONT_H_
#define _GFXFONT_H_

/// Font data stored PER GLYPH
typedef struct {
	uint16_t bitmapOffset;     ///< Pointer into GFXfont->bitmap
	uint8_t  width;            ///< Bitmap dimensions in pixels
    uint8_t  height;           ///< Bitmap dimensions in pixels
	uint8_t  xAdvance;         ///< Distance to advance cursor (x axis)
	int8_t   xOffset;          ///< X dist from cursor pos to UL corner
    int8_t   yOffset;          ///< Y dist from cursor pos to UL corner
} GFXglyph;

/// Data stored for FONT AS A WHOLE
typedef struct { 
	uint8_t  *bitmap;      ///< Glyph bitmaps, concatenated
	GFXglyph *glyph;       ///< Glyph array
	uint8_t   first;       ///< ASCII extents (first char)
    uint8_t   last;        ///< ASCII extents (last char)
	uint8_t   yAdvance;    ///< Newline distance (y axis)
} GFXfont;

#endif // _GFXFONT_H_
```
So hopefully we keep in sync with future changes as well.

closed 2022-02-14 by KurtE