adafruit/Adafruit-GFX-Library #310: GFXcanvas1 custom text compiling issue

orhanyor, 2020-06-30:
Board and screen: Feather M4 , 1.3 ST7789
IDE version: 1.8.13
Basically i'm trying to test this feature from the website https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts the part about the GFX canvas1.

I kept the code at bare minimum but i get errors but also tried following code without this line:

tft.drawBitmap(10, 10, canvas, 128, 32, ST77XX_WHITE, ST77XX_BLACK);

it compiled and i got a hello text on the screen so connections are ok.

This is the test code

```
#include <Adafruit_GFX.h>
#include <Adafruit_MonoOLED.h>
#include <Adafruit_SPITFT.h>
#include <Adafruit_SPITFT_Macros.h>
#include <gfxfont.h>

#include <Adafruit_ST7735.h>
#include <Adafruit_ST7789.h>
#include <Adafruit_ST77xx.h>

#define TFT_DC  10
#define TFT_CS  12
#define TFT_RST 11

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);

GFXcanvas1 canvas(128, 32);

void setup() {

  tft.init(240, 240); 
  tft.fillScreen(ST77XX_BLACK);

  /* this part is to test if connections are ok and see if the screen actually works */
  tft.setCursor(0, 0); 
  tft.setTextColor(ST77XX_WHITE); 
  tft.setTextSize(3);  
  tft.println("Hello"); 
}

void loop() {

canvas.println("I like cake");
tft.drawBitmap(10, 10, canvas, 128, 32, ST77XX_WHITE, ST77XX_BLACK);

}
```

sadly i get this error:

```
AppData\Local\Temp\arduino_modified_sketch_869866\canvas_test.ino: In function 'void loop()':
canvas_test:41:67: error: no matching function for call to 'Adafruit_ST7789::drawBitmap(int, int, GFXcanvas1&, int, int, int, int)'
 tft.drawBitmap(10, 10, canvas, 128, 32, ST77XX_WHITE, ST77XX_BLACK);
                                                                   ^
AppData\Local\Temp\arduino_modified_sketch_869866\canvas_test.ino:1:0:
Documents\Arduino\libraries\Adafruit-GFX-Library/Adafruit_GFX.h:81:8: note: candidate: void Adafruit_GFX::drawBitmap(int16_t, int16_t, const uint8_t*, int16_t, int16_t, uint16_t)
   void drawBitmap(int16_t x, int16_t y, const uint8_t bitmap[], int16_t w,
        ^~~~~~~~~~
Documents\Arduino\libraries\Adafruit-GFX-Library/Adafruit_GFX.h:81:8: note:   candidate expects 6 arguments, 7 provided
Documents\Arduino\libraries\Adafruit-GFX-Library/Adafruit_GFX.h:83:8: note: candidate: void Adafruit_GFX::drawBitmap(int16_t, int16_t, const uint8_t*, int16_t, int16_t, uint16_t, uint16_t)
   void drawBitmap(int16_t x, int16_t y, const uint8_t bitmap[], int16_t w,
        ^~~~~~~~~~
Documents\Arduino\libraries\Adafruit-GFX-Library/Adafruit_GFX.h:83:8: note:   no known conversion for argument 3 from 'GFXcanvas1' to 'const uint8_t* {aka const unsigned char*}'
Documents\Arduino\libraries\Adafruit-GFX-Library/Adafruit_GFX.h:85:8: note: candidate: void Adafruit_GFX::drawBitmap(int16_t, int16_t, uint8_t*, int16_t, int16_t, uint16_t)
   void drawBitmap(int16_t x, int16_t y, uint8_t *bitmap, int16_t w, int16_t h,
        ^~~~~~~~~~
Documents\Arduino\libraries\Adafruit-GFX-Library/Adafruit_GFX.h:85:8: note:   candidate expects 6 arguments, 7 provided
Documents\Arduino\libraries\Adafruit-GFX-Library/Adafruit_GFX.h:87:8: note: candidate: void Adafruit_GFX::drawBitmap(int16_t, int16_t, uint8_t*, int16_t, int16_t, uint16_t, uint16_t)
   void drawBitmap(int16_t x, int16_t y, uint8_t *bitmap, int16_t w, int16_t h,
        ^~~~~~~~~~
Documents\Arduino\libraries\Adafruit-GFX-Library/Adafruit_GFX.h:87:8: note:   no known conversion for argument 3 from 'GFXcanvas1' to 'uint8_t* {aka unsigned char*}'
Using library Adafruit-GFX-Library at version 1.9.0 in folder: Documents\Arduino\libraries\Adafruit-GFX-Library 
Using library Adafruit_BusIO at version 1.3.2 in folder: Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 1.0 in folder: AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.14\libraries\Wire 
Using library SPI at version 1.0 in folder: AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.14\libraries\SPI 
Using library Adafruit_Zero_DMA_Library at version 1.0.8 in folder: Documents\Arduino\libraries\Adafruit_Zero_DMA_Library 
Using library Adafruit-ST7735-Library at version 1.5.17 in folder: Documents\Arduino\libraries\Adafruit-ST7735-Library 
exit status 1
no matching function for call to 'Adafruit_ST7789::drawBitmap(int, int, GFXcanvas1&, int, int, int, int)'

```
Could it be library related or im doing something wrong?

BillyDonahue, 2020-06-30:
It looks like the guide needs to be corrected. A `GFXcanvas1` isn't a bitmap. There must be a call to `.getBuffer()` to pull the bitmap out. So it should have read:

    tft.drawBitmap(x, y, canvas.getBuffer(), 128, 32, foreground,
                   background); // Copy to screen

Can you try that?

orhanyor, 2020-06-30:
> It looks like the guide needs to be corrected. A `GFXcanvas1` isn't a bitmap. There must be a call to `.getBuffer()` to pull the bitmap out. So it should have read:
>
> ```
> tft.drawBitmap(x, y, [...]

Yes! thank you that made it work! its not straight forward tho. i tried to print a simple variable to see if a custom font will flicker and result was abit disappointing but one line of code fixed it.

this is the bit that i modified along with your suggestion.

```
 Variable1++;  // Increase variable by 1
  if(Variable1 > 80) 
  {
    Variable1 = 0;  // Set Variable1 to 0
  }

canvas.setCursor(0, 45);
canvas.setFont(&Retro25);

canvas.setTextSize(0);  

canvas.println(Variable1);
tft.drawBitmap(20, 50, canvas.getBuffer(), 100, 50, ST77XX_WHITE, ST77XX_BLACK);

```
if i upload this code printed values becomes mushy just like as if im not using a canvas route. but if i add this line of code 
canvas.fillScreen(ST77XX_BLACK); just before canvas.setCursor(0, 45); values update without flickering. 
im not sure if im doing it right at this point, im just experimenting but i feel like last two arguments in the tft.drawBitmap where we define background and foreground should do the trick but somehow it does not work that way.

fralbo, 2020-11-19:
Ok good, can you please fix the example given on this [page](https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts)?

PaintYourDragon, 2020-11-19:
Oh no! I apologize. Guide has been updated to use canvas.getBuffer() as it should. Thank you for spotting this.

closed 2020-11-19 by PaintYourDragon