adafruit/Adafruit-GFX-Library #311: Degree symbol also shows underscore below itself

Adityakush, 2020-07-05:
- Arduino board: UNO

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.10

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

Hi!
I am using Arduino uno with Nokia 5110 display for showing temperature. In order to display degree symbol after temperature, I use the line
```
  display.write(167);
  display.display();
```
But along with the degree symbol is also displayed a dash (like an underscore), however, this kind of underscore is not displayed along with other characters. 
Here is the initialization code also
```
  display.begin();
  display.setContrast(57);
  display.clearDisplay();
  display.setTextSize(2);
  display.setTextColor(BLACK);
  display.setCursor(0,0);
  display.println("PIDTEMP");
  display.display();
  delay(1000);
```
 I am pretty sure this has been wrong in the library itself. Please correct this. Thanks

BillyDonahue, 2020-07-05:
The degree sign is glyph number 248. Try that.

Glyph number 167 is not a degree sign, it's a `MASCULINE ORDINAL INDICATOR`, a raised underlined `o`.
it follows 166, the `FEMININE ORDINAL INDICATOR`, which is a raised underlined `a`.

Adityakush, 2020-07-06:
Thanks a ton for that info.

closed 2020-07-06 by Adityakush