adafruit/Adafruit-GFX-Library #319: reserved identifier violation

elfring, 2020-10-26:
I would like to point out that identifiers like “[`_ADAFRUIT_GFX_H`](https://github.com/adafruit/Adafruit-GFX-Library/blob/f7db17f3174f32a76ff6329b9656dd4f8667eeff/Adafruit_GFX.h#L1 "Update candidate")” and “[`_GFXFONT_H_`](https://github.com/adafruit/Adafruit-GFX-Library/blob/9b1781b90d06be321e5e176c8120daa783341384/gfxfont.h#L7 "Another update candidate")” [do not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier#DCL51-CPP.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

ladyada, 2020-10-26:
sure, submit a PR!

mikeysklar, 2026-11-06:
Thanks for the report, @elfring.

Other code now depends on both names, so renaming them would break it.

| Name | Used by | What breaks |
|---|---|---|
| `_ADAFRUIT_GFX_H` | ZinggJM/SSD1283A, Stutchbury/DisplayUtils, MHotchin/RLEBitmap | `#ifdef` checks for GFX, so GFX support turns off silently |
| `_GFXFONT_H_` | Copies of `gfxfont.h` in M5Stack, Control-Surface, TTGO_TWatch_Library | Shared guard stops `GFXfont` being defined twice |

Closing this one.

closed 2026-11-06 by mikeysklar

elfring, 2026-11-06:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/bugprone/reserved-identifier.html "Checking the application of reserved identifiers")?