adafruit/Adafruit-GFX-Library #320: Remove unnecessary null pointer checks

elfring, 2020-10-26:
[Extra null pointer checks are not needed](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") in [the destructor for the class “Adafruit_GrayOLED”](https://github.com/adafruit/Adafruit-GFX-Library/blob/d9220ba7d3cb49915f6aca0feef680e3fed0c400/Adafruit_GrayOLED.cpp#L149 "Update candidate").

mikeysklar, 2026-10-30:
Agreed, delete and free() are safe on NULL.
The checks are harmless and match the other Adafruit display libraries.
We'll leave them as is to avoid churn.

I used Claude Code to check this.

closed 2026-10-30 by mikeysklar

elfring, 2026-10-30:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability/delete-null-pointer.html "Deletion of unnecessary null pointer checks")?