adafruit/Adafruit-GFX-Library #322: BIG FONT

chiehun, 2020-11-07:
Hi,

to have some hangeul scharacters I used the fontconvert tool (Ubuntu , minGW, GodoM following fontconvert_win.md) to convert GodoMfont
한글 아래 환경에 맞추어서 9pt를 
How ever, results do work but look very ugly.
결과가 엉망이었다,
 They look as if their width does not match and therefore are jolted (or compressed/forged - not sure how to translate).
Because the index number does not match, the letters are pushed out.

How can I print the correct text?
{ 65266,  10,  15,  12,    1,  -12 },   // 0x93EF
  { 65285,  10,  15,  12,    1,  -12 },   // 0x93F0
  { 65304,  10,  15,  12,    1,  -12 },   // 0x93F1
  { 65323,  10,  15,  12,    1,  -12 },   // 0x93F2
  { 65342,  10,  15,  12,    1,  -12 },   // 0x93F3
  { 65361,  10,  15,  12,    1,  -12 },   // 0x93F4
  { 65380,  10,  15,  12,    1,  -12 },   // 0x93F5
  { 65399,  10,  15,  12,    1,  -12 },   // 0x93F6
  { 65418,  10,  15,  12,    1,  -12 },   // 0x93F7
  { 65437,  10,  15,  12,    1,  -12 },   // 0x93F8
  { 65456,  10,  15,  12,    1,  -12 },   // 0x93F9
  { 65475,  10,  15,  12,    1,  -12 },   // 0x93FA
  { 65494,  10,  15,  12,    1,  -12 },   // 0x93FB
  { 65513,  10,  15,  12,    1,  -12 },   // 0x93FC
  { 65532,  10,  15,  12,    1,  -12 },   // 0x93FD
  {    15,  10,  15,  12,    1,  -12 },   // 0x93FE
  {    34,  10,  15,  12,    1,  -12 },   // 0x93FF
  {    53,  10,  15,  12,    1,  -12 },   // 0x9400
  {    72,  10,  15,  12,    1,  -12 },   // 0x9401
  {    91,  10,  15,  12,    1,  -12 },   // 0x9402
  {   110,  10,  15,  12,    1,  -12 },   // 0x9403
  {   129,  10,  15,  12,    1,  -12 },   // 0x9404
  {   148,  10,  15,  12,    1,  -12 },   // 0x9405
  {   167,  10,  15,  12,    1,  -12 },   // 0x9406
  {   186,  10,  15,  12,    1,  -12 },   // 0x9407
  {   205,  10,  15,  12,    1,  -12 },   // 0x9408
  {   224,  10,  15,  12,    1,  -12 },   // 0x9409
  {   243,  10,  15,  12,    1,  -12 },   // 0x940A
  {   262,  10,  15,  12,    1,  -12 },   // 0x940B
  {   281,  10,  15,  12,    1,  -12 },   // 0x940C
  {   300,  10,  15,  12,    1,  -12 },   // 0x940D
  {   319,  10,  15,  12,    1,  -12 },   // 0x940E
  {   338,  10,  15,  12,    1,  -12 },   // 0x940F
  {   357,  10,  15,  12,    1,  -12 },   // 0x9410
  {   376,  10,  15,  12,    1,  -12 },   // 0x9411
  {   395,  10,  15,  12,    1,  -12 },   // 0x9412
  {   414,  10,  15,  12,    1,  -12 },   // 0x9413

chiehun, 2020-11-07:
> Hi,
> 
> to have some hangeul scharacters I used the fontconvert tool (Ubuntu , minGW, GodoM following fontconvert_win.md) to convert GodoMfont

> How ever, results do work but look very ugly.

> They look as if their width does not match and therefore are jolted (or compressed/forged - not sure how to translate).
> Because the index number does not match, the letters are pushed out.
> 
> How can I print the correct text?

> { 65266, 10, 15, 12, 1, -12 }, // 0x93EF
> { 65285, 10, 15, 12, 1, -12 }, // 0x93F0
> { 65304, 10, 15, 12, 1, -12 }, // 0x93F1
> { 65323, 10, 15, 12, 1, -12 }, // 0x93F2
> { 65342, 10, 15, 12, 1, -12 }, // [...]

ejiyuan, 2024-11-21:
up

closed 2026-10-30 by mikeysklar