adafruit/Adafruit-GFX-Library #327: getTextBounds returning an incorrect width

mrfaptastic, 2020-12-08:
Refer to: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/43

Specifically, https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/43#issuecomment-740575875

Naios, 2021-01-18:
I can confirm the issue. With the default font and default size the text bounds are always returned with an additional pixel in width and height which can also be seen in this image (the text "Huhu" is rasterized using AdafruitGFX and the preferred size is exactly what `getTextBounds` returns):

[image]

mikeysklar, 2026-11-09:
This is working as intended. With the default font, `getTextBounds()` returns whole 6x8 character cells, including the blank column and row that space characters apart. That is the same area `drawChar()` paints when you set a background color, so the box matches what gets drawn and can be used to clear or right-align text.

Tested on a Feather HUZZAH ESP8266 with the 3.5" HX8357 FeatherWing on current master. Text is drawn with a blue background, and the red ring sits just outside the returned bounds:

[image]

Changing it now would shift every layout that relies on these numbers, so closing. Please reopen if you see a case where the box does not match the painted cells.

closed 2026-11-09 by mikeysklar