adafruit/Adafruit-GFX-Library #329: ESP32 DMA support

Avamander, 2021-01-01:
I took a peek and it seems that there's already SAMD MCU DMA support, but not ESP32. Are there any plans on adding ESP32 DMA support to Adafruit GFX?

ladyada, 2021-01-01:
if someone submits a PR, we'd definitely review it for addition!

mikeysklar, 2026-11-14:
Measured how close the current non-DMA code gets to the SPI wire speed on ESP32.

Feather ESP32-S3 TFT (240x135 ST7789), esp32 core 3.3.8, ms per full frame:

| SPI clock | Wire time | `fillScreen()` | `writePixels()` frame |
|---|---|---|---|
| 20 MHz | 25.9 | 32.0 | 31.7 |
| 40 MHz | 13.0 | 19.0 | 18.7 |
| 80 MHz | 6.5 | 12.5 | 12.2 |

[image]

Every frame costs about 6 ms more than the wire time, at any clock.

That is the CPU refilling the SPI FIFO between small chunks, and the CPU is busy the whole time.

At 80 MHz DMA could nearly halve the frame time and free the CPU while it draws.

So this is worth doing, and a PR is welcome as ladyada said.

<details>
<summary>Test sketch</summary>

```cpp
// #329: how close does Adafruit_SPITFT get to SPI wire speed on ESP32 (no DMA)?
// Feather ESP32-S3 TFT, built-in 240x135 ST7789.
#include <Adafruit_GFX.h>
#include <Adafruit_ST7789.h>
#include <SPI.h>

Adafruit_ST7789 tft(TFT_CS, TFT_DC, TFT_RST);
uint16_t buf[240 * 15];
String logText;

void run(uint32_t hz) {
  tft.setSPISpeed(hz);
  const uint32_t px = 240UL * 135;
  float wire = px * 16.0f / hz * 1000.0f; // ms if the bus never idles

  const uint16_t colors[] = {0xF800, 0x07E0, 0x001F, 0xFFFF, 0x0000};
  uint32_t t = micros();
  for (int i = 0; i < 20; i++)
    tft.fillScreen(colors[i % 5]);
  float fill = (micros() - t) / 20000.0f;

  t = micros();
  for (int n = 0; n < 20; n++) {
    tft.startWrite();
    tft.setAddrWindow(0, 0, 240, 135);
    for (int i = 0; i < 9; i++)
      tft.writePixels(buf, 240 * 15);
    tft.endWrite();
  }
  float frame = (micros() - t) / 20000.0f;

  char line[40];
  snprintf(line, sizeof line, "%2lu %5.1f %5.1f %5.1f\n", hz / 1000000, wire, fill, frame);
  logText += line;
}

void setup() {
  Serial.begin(115200);
  pinMode(TFT_BACKLITE, OUTPUT);
  digitalWrite(TFT_BACKLITE, HIGH);
  pinMode(TFT_I2C_POWER, OUTPUT);
  digitalWrite(TFT_I2C_POWER, HIGH);
  tft.init(135, 240);
  tft.setRotation(3);
  for (int i = 0; i < 240 * 15; i++)
    buf[i] = ((i % 240) / 30 + (i / 240 / 5)) & 1 ? 0xFFE0 : 0x001F;

  logText = "MHz wire  fill frame\n";
  run(20000000);
  run(40000000);
  run(80000000);

  tft.fillScreen(0);
  tft.setTextSize(2);
  tft.setTextColor(0xFFFF);
  tft.setCursor(0, 20);
  tft.print(logText);
  tft.setCursor(0, 110);
  tft.print("ms per full frame");
}

void loop() {
  Serial.print(logText);
  Serial.println("--");
  delay(3000);
}
```

</details>

closed 2026-11-15 by mikeysklar