adafruit/Adafruit-GFX-Library #339: Contrast Issue Heltec 1.54

nnc6, 2021-03-05:
Hello, I'm having some contrast troubles with this display. As the display is updating contrast is ok but one second later it becomes weaker. Below two pictures to make myself understood. Trying different things I found that this is caused by 'Busy' connection: if I rapidly remove the connection suddendly after updating of the display contrast remains ok and the display updating process ends up correctly. The problem is that without this connection it's not updating anymore, I also tried to set to -1 the following line but it's not updating too. GxEPD_Class display(io, /*RST=D4*/ -1, /*BUSY=D2*/ D6); // no RST, no BUSY
I'm using it on a Wemos D1 Mini. 
Can somebody help me to solve this problem?
Below the full 
`#include <GxEPD.h>

#include <GxGDEH0154D67/GxGDEH0154D67.h>  // 1.54" b/w

#include <BitmapExamples.h>

// FreeFonts from Adafruit_GFX
#include <Fonts/FreeMono9pt7b.h>
#include <Fonts/FreeMonoBold9pt7b.h>
#include <Fonts/FreeMonoBold12pt7b.h>
#include <Fonts/FreeMonoBold18pt7b.h>
#include <Fonts/FreeMonoBold24pt7b.h>
#include <Fonts/FreeSansOblique24pt7b.h>
#include <Fonts/chubbybrush30pt7b.h>
#include <Fonts/orange30pt7b.h>
#include <Fonts/orange50pt7b.h>
#include <Fonts/mrsmonsterboldout30pt7b.h>
#include <Fonts/mrsmonsterdd30pt7b.h>
#include <Fonts/mrsmonster30pt7b.h>
#include <Fonts/ConcreteFantasia30pt7b.h>
//#include <Fonts/.h>

#include <GxIO/GxIO_SPI/GxIO_SPI.h>
#include <GxIO/GxIO.h>

// for SPI pin definitions see e.g.:
// C:\Users\xxx\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\generic\common.h

GxIO_Class io(SPI, /*CS=D8*/ D8, /*DC=D3*/ D3, /*RST=D4*/ D6); // arbitrary selection of D3(=0), D4(=2), selected for default of GxEPD_Class
//GxEPD_Class display(io, /*RST=D4*/ 2, /*BUSY=D2*/ 4); // default selection of D4(=2), D2(=4)
// Heltec E-Paper 1.54" b/w without RST, BUSY
GxEPD_Class display(io, /*RST=D4*/ -1, /*BUSY=D2*/ D6); // no RST, no BUSY

#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

#define BME280_ADDRESS 0x76

Adafruit_BME280 bme;

void setup()
{
  pinMode(D6, OUTPUT);
  pinMode(D0,OUTPUT);
  digitalWrite (D0,LOW);
  delay(2000);
  if (!bme.begin(BME280_ADDRESS)) {
    //    digitalWrite (D5,HIGH);
    Serial.println("Could not find a valid BME280 sensor, check wiring or BME-280 address!");
    while (1)
      bme.setSampling(Adafruit_BME280::MODE_FORCED,
                      Adafruit_BME280::SAMPLING_X1, // temperature
                      Adafruit_BME280::SAMPLING_X1, // pressure
                      Adafruit_BME280::SAMPLING_X1, // humidity
                      Adafruit_BME280::FILTER_OFF);
  }
  Serial.begin(115200);
  Serial.println();
  Serial.println("setup");

  display.init(115200); // enable diagnostic output on Serial

  Serial.println("setup done");
 
}

void loop()
{
  delay(1000);
  display.drawPaged(showFontCallback);
  delay(600000);
}

void showFontCallback()
{
  
  // Reading BME280 sensor data
  bme.takeForcedMeasurement(); // has no effect in normal mode
  const char* name = "Climatica";
  const GFXfont* f = &orange50pt7b;
  const GFXfont* g = &FreeMonoBold12pt7b;
  display.setRotation(1);
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(g);
  display.setCursor(0, 90);
//  display.println();
//  display.println(name);
  display.setFont(f);
//  display.println();
  display.print(bme.readTemperature(),1);
  display.setFont(g);
  display.println("C");
  display.setCursor(0, 180);  
  display.setFont(f);
  display.print(bme.readHumidity(),0);
  display.setFont(g);
  display.println(" %");
  digitalWrite(D6, HIGH);
}
`

nnc6, 2021-03-05:
[image]
[image]

ladyada, 2021-03-05:
please contact the TFT vendor for help

closed 2021-03-05 by ladyada