adafruit/Adafruit-GFX-Library #353: CRC Error for actual update from 1.10.7 to 1.10.9 ?

electronic-cosmos, 2021-05-26:
CRC error shown for library update.
Seems to jump over 1.10.8
Please check your data.

caternuson, 2021-05-27:
related:
https://forums.adafruit.com/viewtopic.php?f=25&t=179714

ladyada, 2021-05-27:
@per1234 any ideas? everything looks ok :/

caternuson, 2021-05-27:
also possibly related:
https://forums.adafruit.com/viewtopic.php?f=52&t=179736
(issue may be more generic than this specific repo)

electronic-cosmos, 2021-05-27:
Here my screen for your better understanding.
Each time I start Arduino IDE hint for updating library comes up by the same result with CRC error!
[image]

electronic-cosmos, 2021-05-27:
Test I did right no:
1. removed (deleted) GFX by hand
2. reinstalled 1.10.9 -> CRC error
3. reinstalled 1.10.8 -> CRC error
4. shocked :-( reinstalled 1.10.7 -> normal installation :-)

Best regards ,
Michael

HA4ever37, 2021-05-28:
Same here I'm getting the CRC Error. I am able to install other libraries without any issues.

ladyada, 2021-05-28:
@cmaglie hihi i know the lib manager was down for a few hours, could it have mis-crc'd some libraries during that time?

per1234, 2021-05-28:
Hi all. I apologize for this. The problem is something in the Arduino Library Manager back end, nothing caused by anything Adafruit has done or can do anything to fix.

We will investigate this today and I'll provide updates as soon as I know anything more.

per1234, 2021-05-28:
Hello, I'm back with good news. The incorrect checksums in the Arduino Library Manager index for the Adafruit libraries have now been corrected.

Since they were the ones reported, the developers gave priority to an immediate fix for these libraries.  There are some other non-Adafruit libraries that were also affected and the IDE's library index cache timer hasn't yet expired for the fix on those. So you might still find a few others with the CRC mismatch, but those should be resolved automatically in a few hours when the cache expiration causes the IDE to pull the new index.

Please let me know if you encounter any issues.

ladyada, 2021-05-28:
huzzah thank you both :) its amazing how reliable the lib manager is that this has only happened once!

closed 2021-05-28 by ladyada

caternuson, 2021-05-28:
Awesome. Thanks @per1234 !