adafruit/Adafruit-GFX-Library #373: fontconvert upgrades

ftlpilot, 2021-11-22:
Can I put in a request to include the degree character 0xB0 (176 dec) glyph when  fontconvert converts a TTF file.

Also, is it possible to fix the main algorithm, so odd dots don't sometimes appear at the edges of large fonts.

BillyDonahue, 2021-11-23:
Can you run `fontconvert` yourself with a range that includes the 0xB0 code point?

The problem is that the GFXfont struct has no way to represent a noncontiguous range of code points, and takes the reasonable default of only encoding the ASCII printable characters for the built-in fonts, but you aren't constrained to that range as a user of the library. You can generate and use your own `GFXfont`.

PS: What odd dots? I have been working with `GFXfont` a lot and haven't noticed this problem.

virxkane, 2021-11-25:
> The problem is that the GFXfont struct has no way to represent a noncontiguous range of code points

This can be changed, for examples, like this: https://github.com/virxkane/nucleo-f446re-st7735spi-shield/blob/main/Inc/gfxfont.h
But, unfortunatly, this is out of [roadmap](https://github.com/adafruit/Adafruit-GFX-Library#roadmap).

ftlpilot, 2021-11-25:
Thankx – useful info, I’ve setup a spare computer with Linux Mint, so I’ll see what I can get working with a modified fontconvert. 

 

I need LARGE (80pt) fonts for clocks, but only 0-9 and colon, to minimise progmem usage – if possible.

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

BillyDonahue, 2021-11-26:
Happy hacking. If you're more handy with Python than C, I've ported fontconvert to Python. It also includes a `gfxfont_preview` tool to annotate a font header with C++ comments showing each glyph in ASCII art, which I find really useful for seeing what you're doing.

https://github.com/BillyDonahue/Adafruit-GFX-Library/tree/fontconvert_py

nilclass, 2022-07-31:
@ftlpilot I was also looking for a way to include the degree character (`°`) into my font, and ended up hacking fontconvert to allow me to do it. Instead of adding support for multiple ranges (which complicates things not only on fontconvert's side, but also requires changes to the library code that uses the font), I decided to instead include the glyphs I needed in place of characters I did not need.
For more details, see this PR: https://github.com/adafruit/Adafruit-GFX-Library/pull/395

mikeysklar, 2026-11-09:
For the degree sign: the browser fontconvert can do it now, https://adafruit.github.io/Adafruit-GFX-Library/fontconvert/
Set last char to 176 and print `"\xB0"`. Example and photo in #452.
The stray dots on large fonts are still open.

mikeysklar, 2026-11-14:
I couldn't reproduce the stray dots with the current fontconvert.

I converted 0-9 and `:` from 10 fonts and checked every glyph for specks of 6 pixels or fewer.

| Fonts | Sizes | Hinting | Specks |
|---|---|---|---|
| Arial, Arial Bold, Georgia, Times New Roman, Verdana, Trebuchet MS, Tahoma, Impact, Courier New, Comic Sans MS | 60, 80 pt | default and `-a` | 0 |

Arial 80 pt and Georgia 60 pt on a Feather HUZZAH ESP8266 with the 3.5" HX8357 FeatherWing:

[image]

The degree sign was covered above, so closing this. If you still see dots, please reopen with the font and size.

closed 2026-11-14 by mikeysklar