adafruit/Adafruit-GFX-Library #393: I2C conflict with Adafruit GFX + ST7735 and ESPAsyncWebServer

OslafEman, 2022-07-18:
Arduino IDE 1.8.19
Windows 
ESP32 (MH ET LIVE ESP32minikit)

Hello, in an asynchronous file server (**ESPAsyncWebServer + AsyncTCP**) the combination of **Adafruit GFX and Adafruit ST7735** libraries interfere with the I2C connection of a keyboard. It doesn't matter if you use the **SPIFFS_ImageReader library** (fork of Adafruit_ImageReader) or render images in hexadecimal format, the keyboard has almost zero sensitivity.

I currently use **ESP32WebServer** with the same Adafruit libraries and **Adafruit_ImageReader** with no keyboard issues.

Oslaf

caternuson, 2022-07-18:
Cross posted:
https://forums.adafruit.com/viewtopic.php?f=22&t=192860

mikeysklar, 2026-10-21:
This is being discussed in the forum thread linked above.
The GFX core does not use the I2C bus.
The ST7735 display uses SPI, not I2C.
Closing this one.

closed 2026-10-21 by mikeysklar