adafruit/Adafruit-GFX-Library #400: SPI_BEGIN_TRANSACTION is slowing down transmissions on SAMD51

fvzeppelin, 2022-09-06:
- Arduino board:  Grand Central M4 Express

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.19

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

1. Setup SPITFT with 24MHz SCK (which is hard to set up, s. [https://github.com/arduino/ArduinoCore-samd/issues/680](url). In my case, I used an SSD1331 RGB OLED.
2. Using the current, unmodified GFX lib 1.11.3, I measure a time between consecutive 8-bit transfers of ~6us.
3. Modification in Adafruit_SPITFT.cpp: comment out line 2068:
```
inline void Adafruit_SPITFT::SPI_BEGIN_TRANSACTION(void) {
if (connection == TFT_HARD_SPI) {
#if defined(SPI_HAS_TRANSACTION)
//    hwspi._spi->beginTransaction(hwspi.settings);
#else // No transactions, configure SPI manually...
#if defined(__AVR__) || defined(TEENSYDUINO) || defined(ARDUINO_ARCH_STM32F1)
    hwspi._spi->setClockDivider(SPI_CLOCK_DIV2);
#elif defined(__arm__)
    hwspi._spi->setClockDivider(11);
#elif defined(ESP8266) || defined(ESP32)
    hwspi._spi->setFrequency(hwspi._freq);
#elif defined(RASPI) || defined(ARDUINO_ARCH_STM32F1)
    hwspi._spi->setClock(hwspi._freq);
#endif
    hwspi._spi->setBitOrder(MSBFIRST);
    hwspi._spi->setDataMode(hwspi._mode);
#endif // end !SPI_HAS_TRANSACTION
  }
}
```
This modification will reduce the above mentioned delay between concecutive 8-bit transfers to ~0.8us(!).

As to my understanding of the library's design I think that this behaviour is unintended and, additonally, it is really painful because it makes a huge difference in screen build-up: It's the difference between slow top-to-down build-up and an instant one.

fvzeppelin, 2022-09-09:
Here is the code I used now to reproduce the issue on an ItsyBitsy M4. Just provide rgb data in splash.h and you will see the difference with the line commented out as described above:
```
#define OLED_CS_PIN  A5
#define OLED_DC_PIN  A4
#define OLED_RST_PIN A3

#include <Adafruit_GFX.h>
#include <Adafruit_SSD1331.h>
Adafruit_SSD1331 *oled;
#include "splash.h"
uint16_t OLED_Background_Color   = 0x0000;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  SPI.begin();
  oled = new Adafruit_SSD1331(&SPI, OLED_CS_PIN, OLED_DC_PIN, OLED_RST_PIN);
  oled->begin(24000000);
  sercom1.setBaudrateSPI(0);
  oled->fillScreen(OLED_Background_Color);
  oled->drawRGBBitmap(0, 0, (const uint16_t *)image_data_splash_rgb, 96, 64);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print("PCHCTRL: ");
  Serial.println(GCLK->PCHCTRL[SERCOM1_GCLK_ID_CORE].reg);
  Serial.print("BAUDREG: ");
  Serial.println(SERCOM1->SPI.BAUD.reg);
  delay(1000);
}
```
Actually, I do not understand this behaviour, as the mentioned line *should* not make a difference. Nevetheless, it does.

mikeysklar, 2026-11-14:
Each `SPI.beginTransaction()` on SAMD51 disables the SERCOM, sets it up again and re-enables it, even when nothing changed.

That costs about 4.4 us, and it also undoes a manual `setBaudrateSPI()`.

`drawPixel()` opens a transaction for every pixel, so pixel-by-pixel drawing pays it every time.

Wrap the drawing in `startWrite()` and `endWrite()`, and use the `write...()` calls inside, to pay it once:

```cpp
tft.startWrite();
for (...) tft.writePixel(x, y, color);
tft.endWrite();
```

10,000 pixels on a Metro M4 AirLift Lite at 24 MHz, Adafruit SAMD core 1.7.17, 5 runs each:

| | Time |
|---|---|
| `drawPixel()` | 161,962 us |
| `startWrite()` + `writePixel()` + `endWrite()` | 112,419 us |
| `SPI.beginTransaction()` + `endTransaction()` alone | 44,138 us |

GFX keeps the transaction so a shared SPI bus stays safe. Skipping the reconfigure when the settings match would be a change to ArduinoCore-samd's `SPIClass::beginTransaction()`.

Closing, please reopen if batching doesn't cover your case.

<details>
<summary>Test sketch</summary>

```cpp
// #400: cost of one SPI transaction per call on SAMD51.
// drawPixel() opens and closes a transaction for every pixel; writePixel()
// between startWrite() and endWrite() opens one for the whole batch.
// Metro M4 AirLift Lite, HX8357 driver at 24 MHz, no panel attached
// (timing only).
#include <Adafruit_GFX.h>
#include <Adafruit_HX8357.h>

Adafruit_HX8357 tft(10, 9); // CS, DC, hardware SPI
const int N = 10000;

void setup() {
  Serial.begin(115200);
  while (!Serial && millis() < 4000)
    ;
  tft.begin(24000000);
  SPISettings s(24000000, MSBFIRST, SPI_MODE0);
  for (int run = 0; run < 5; run++) {
    uint32_t t = micros();
    for (int i = 0; i < N; i++)
      tft.drawPixel(i % 320, i / 320, i);
    uint32_t a = micros() - t;

    t = micros();
    tft.startWrite();
    for (int i = 0; i < N; i++)
      tft.writePixel(i % 320, i / 320, i);
    tft.endWrite();
    uint32_t b = micros() - t;

    t = micros();
    for (int i = 0; i < N; i++) {
      SPI.beginTransaction(s);
      SPI.endTransaction();
    }
    uint32_t c = micros() - t;
    Serial.print("#400 run ");
    Serial.print(run);
    Serial.print(": drawPixel ");
    Serial.print(a);
    Serial.print(" us, startWrite+writePixel ");
    Serial.print(b);
    Serial.print(" us, beginTransaction alone ");
    Serial.print(c);
    Serial.println(" us");
  }
}

void loop() {}
```

</details>

closed 2026-11-14 by mikeysklar