adafruit/Adafruit-GFX-Library #407: getTextBounds with Grand Central M4 SAMD51 board not compiling

nickpetersen93, 2022-11-01:
Unable to compile when attempting to upload script to Grand Central M4 Express using the Adafruit 2.8" TFT Touch Shield v2 - Capacitive Touch Shield. All libraries are up to date. Error exclusively involves getTextBounds function. 

Error message:
/Users/np/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:114:8: note: candidate: 'void Adafruit_GFX::getTextBounds(const char*, int16_t, int16_t, int16_t*, int16_t*, uint16_t*, uint16_t*)'
  114 |   void getTextBounds(const char *string, int16_t x, int16_t y, int16_t *x1,
      |        ^~~~~~~~~~~~~
/Users/np/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:114:34: note:   no known conversion for argument 1 from 'String' to 'const char*'
  114 |   void getTextBounds(const char *string, int16_t x, int16_t y, int16_t *x1,
      |                      ~~~~~~~~~~~~^~~~~~
/Users/np/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:116:8: note: candidate: 'void Adafruit_GFX::getTextBounds(const __FlashStringHelper*, int16_t, int16_t, int16_t*, int16_t*, uint16_t*, uint16_t*)'
  116 |   void getTextBounds(const __FlashStringHelper *s, int16_t x, int16_t y,
      |        ^~~~~~~~~~~~~
/Users/np/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:116:49: note:   no known conversion for argument 1 from 'String' to 'const __FlashStringHelper*'
  116 |   void getTextBounds(const __FlashStringHelper *s, int16_t x, int16_t y,
      |                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/Users/np/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:118:8: note: candidate: 'void Adafruit_GFX::getTextBounds(const String&, int16_t, int16_t, int16_t*, int16_t*, uint16_t*, uint16_t*)'
  118 |   void getTextBounds(const String &str, int16_t x, int16_t y, int16_t *x1,
      |        ^~~~~~~~~~~~~
/Users/np/Documents/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:118:72: note:   no known conversion for argument 4 from 'int*' to 'int16_t*' {aka 'short int*'}
  118 |   void getTextBounds(const String &str, int16_t x, int16_t y, int16_t *x1,
      |                                                               ~~~~~~~~~^~
no matching function for call to 'Adafruit_ILI9341::getTextBounds(String&, int, int, int*, int*, int*, int*)'

BillyDonahue, 2022-11-01:
It kind of looks like you are passing the wrong type of integers to `getTextBounds`.

GFX uses `int16_t` for screen positions, not `int`.

The error message has been distorted quite a bit here. It's all one one of text with stripped spaces. A cleaner dump would help. It looks like it's missing the part of the error message that would show where `getTextBounds` was invoked by user code. Are you calling `getTextBounds` in your `.ino` sketch?

nickpetersen93, 2022-11-02:
That fixed this issue! I have been using the same script on an Arduino Mega with no problems (just some permissive warnings), but I changed the variable types on that original script as well to avoid future issues.